Ultimaker/Cura #9389: [4.8.0] Tree support ignores areas requiring support

war4peace, 2021-03-12:
**Application version**
Cura 4.8.0, Arachne Engine Beta

**Platform**
Windows 10 X64, 32 GB DDR4, CPU Intel i7 6800K, 2x 1080 Ti in SLI

**Printer**
Creality CR 10 MAX with modifications (Direct Drive, etc)

**Reproduction steps**
1. Bought a Yoda model from CGTrader (https://www.cgtrader.com/3d-models/miniatures/figurines/yoda-sculpture-3d-print-model)
2. Resized to 200mm Z height (locked axes)
3. Sliced using Tree Supports option, his left hand (the one holding the cane), palm area is never supported, as if the algorithm completely ignores it. Also, Yoda left ear is ignored when 

**Screenshot(s)**
Model showing problematic area (4.8.0):
[image]

Left hand claws shown being printed in mid-air (4.8.0):
[image]

First layer of left ear, shows as being printed horizontally (parallel to ground) but no support shown (4.8.0)
[image]

Model after slicing, showing lack of supports (4.8.0)
[image]

Same issues in Arachne Beta:
[image]

**Actual results**
No supports in area clearly requiring supports

**Expected results**
Supports in that area.

**Project file**
This is a paid model, I don't think it would be wise to share it publicly (breach of license).
I can, however, transfer the project to a representative privately, for debugging purposes.

**Log file**
Not applicable

**Additional information**
I have tried "Support Overhang Angle" values starting at 65 and dropping all the way to 40, with the same result. Tree support algorithm ignores areas where support is absolutely required.

fvrmr, 2021-03-12:
Hi @war4peace thank you for your report.
I just sliced a familiar model (found on Thingiverse) and it wasn't a problem for me. 

[image]

So I am really curious to your project file. I saw that you've posted this also on a Facebook group. I will contact you there.

war4peace, 2021-03-12:
@fvrmr Hi, There are some differences between how we were slicing. I have not chosen "Everywhere" under "Support Placement" because the model doesn't seem to require it, and the last thing I want is for supports to affect the model's upper surfaces.
At any rate, I have sent you a link to the model on Facebook Messenger.

fvrmr, 2021-03-16:
It looks like this has to do with the Z distance and this is already a known bug, see here: https://github.com/Ultimaker/Cura/issues/8566

The work around I found for this model is changing the the `support x/y distance` to 0.4 mm the `support overhang angle` to 45 degrees and the `tree support collision resolution` to 0.4mm. Then I added a support blocker to the places that don't need support. 

An other solution is add custom support to the places there is no support created. 

I will close this issue since it is a duplicate. But feel free to reach out for more questions.

closed 2021-03-16 by fvrmr