Ultimaker/Cura #9390: [FR] Supports : variable line width / flow

yet-another-average-joe, 2021-03-12:
### Is your feature request related to a problem?

When printing PETG, it is better to print supports with a reduced line width (true for large nozzles like 1.0 or 1.2mm, but also for small ones - 0.4mm). For example, with a 0.5mm nozzle, I set supports flow to 50% (= 0.25mm). This makes supports much easier to remove. But sometimes, thin supports do not adhere well to the bed.
(for supports, I play with flow, never with line width)

### Describe the solution you'd like

One more parameter would be interesting : initial flow or line width for supports. 

### Describe alternatives you've considered

Brims, but it they can be difficult to remove...

### Affected users and/or printers

All printers

This is more about supports that are "inside", like supports for counterbores and countersunks for example.

Hope I didn't miss a parameter that already offers this feature !

fvrmr, 2021-03-15:
Hi @yet-another-average-joe thank you for your feature request. 
I will discuss this with the team. Keep you posted!

Ghostkeeper, 2021-03-23:
This issue has been discussed with the team. We consider this feature to not really have a big enough use case to give this priority. We're deferring it to make it clear that we'll probably not get around to this sort of thing soon.

The reason why we don't consider there to be a big use case is that if there are issues with the bed adhesion of support, similar issues should be present for normal parts as well. This depends on the shape of that object and the shape of the support though. It could be that your support is really small and your object big enough to have decent contact area with the bed. But the issue is more with small objects in general than specifically with support.

There are also a number of workarounds you can use:
* You can increase the initial layer line width and then reduce it again in the per-object settings for your models. Tedious, but it gives the precise result you are asking for.
* Instead of reducing the line width for support in general, use a support interface with reduced line width so that the support doesn't stick too much to the object but most of the support still gets printed with a strong structure.
* Use something like Conical Support with a negative angle to increase the surface area of support. I wouldn't recommend this though since it significantly increases the printing time and material usage.