Ultimaker/Cura #9393: Selective Infill

evdtempel, 2021-03-13:
### Application Version

4.8.0 & arachne-beta

### Platform

OS X

### Printer

Prusa i3 MK3/MK3S

### Reproduction steps

I’m seeing issues with a trick that is used to have the slicer create additional strength in certain parts of the model.

The trick is to create 0.101x0.101mm hole in the print where strengthening is desired. This should cause the slicer to generate perimeters around it, instead of creating infill.

Example in PrusaSlicer: https://user-images.githubusercontent.com/1208705/111030121-f201c200-8400-11eb-9cec-ff986b378fba.mov
Example in Cura 4.8:  https://user-images.githubusercontent.com/1208705/111030019-48223580-8400-11eb-9e75-8df8389999cc.mov
Example in Cura Arachne (beta): https://user-images.githubusercontent.com/1208705/111030025-4e181680-8400-11eb-9bc5-b474569af660.mov

When using a slightly bigger hole of 0.201x0.301mm, I can see it starting to have an effect - but it is quite inconsistent.

Example in Cura 4.8:  https://user-images.githubusercontent.com/1208705/111030023-4c4e5300-8400-11eb-8f62-d9781d18440e.mov
Example in Cura Arachne (beta): https://user-images.githubusercontent.com/1208705/111030026-4f494380-8400-11eb-9e2d-e7c377d6c7d9.mov

### Actual results

See rep-steps.

### Expected results

Cura detects the 0.101x0.101mm hole. 
Cura slices the hole correctly in every layer.

4.8-101x101 - https://user-images.githubusercontent.com/1208705/111030019-48223580-8400-11eb-9e75-8df8389999cc.mov
4.8-201x301 - https://user-images.githubusercontent.com/1208705/111030023-4c4e5300-8400-11eb-8f62-d9781d18440e.mov
arachne-101x101 - https://user-images.githubusercontent.com/1208705/111030025-4e181680-8400-11eb-9bc5-b474569af660.mov
arachne-201x301 - https://user-images.githubusercontent.com/1208705/111030026-4f494380-8400-11eb-9e2d-e7c377d6c7d9.mov
prusaslicer-101x101 - https://user-images.githubusercontent.com/1208705/111030121-f201c200-8400-11eb-9cec-ff986b378fba.mov

Cura 3mf-files: [3mfs.zip](https://github.com/Ultimaker/Cura/files/6134767/3mfs.zip)

fvrmr, 2021-03-15:
Hi @evdtempel thank you for your report. I really appreciated the videos.
In Cura you can use the `minimum polygon circumference` and change it to 0.1mm.

**4.8:**
[image]

**Arachne beta:**
[image]

Hope this works for you. Let me know when you have more questions.

evdtempel, 2021-03-16:
@fvrmr yep, thanks, changing `minimum polygon circumference` to 0.1 solves any issues in arachne, working as expected.

4.8 is however showing some weird behaviour.

No top/bottom-layers for the hole:
[image]

The hole is somehow twisted, on certain layers:
[image]

fvrmr, 2021-03-19:
@evdtempel thanks, if I look at it again I have the same
But since Arachne fixes this, we will not fix this in 4.8. Hope you understand!

Ghostkeeper, 2021-03-23:
Actually, the skin disappearing is expected behaviour. It's done for the Skin Removal Width setting which removes small patches of skin which are usually not contributing to the strength of the part. The article of the [Settings Guide](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/infill/skin_preshrink.md) about this setting explains a common pitfall when the Skin Removal Width approaches the wall width, that there can then be gaps of infill exposed to the air. That is what you're seeing here, essentially (although the "air" is just a really tiny air pocket).

This Skin Removal Width behaves slightly differently in Arachne because it used to be based on the shape of the outer wall and it's then based on the shape of the skin/infill area instead due to the way variable line width inherently misaligns things when compared to the simple offset of Skin Removal Width. But the pitfall is still there, effectively.

I will close this issue because both of the issues raised here are expected behaviour. They are the results of the default settings, for which trade-offs have been made by the people who made those defaults. The Settings Guide explains most of those trade-offs, if you're interested.

closed 2021-03-23 by Ghostkeeper