Ultimaker/Cura #9394: Arachne Beta - Slicing Hangs

GregValiant, 2021-03-13:
### Application Version

Arachne Beta

### Platform

Win 10

### Printer

Creality

### Reproduction steps

Opened the model
Tried to slice the model

### Actual results

Slicing did not complete.  Slider went blank.

### Expected results

Should have finished slicing

Load the 3mf and try to slice.  I couldn't tell what setting (or combination) caused the problem.
The project file and Cura.log are in the zip file.
[cura.zip](https://github.com/Ultimaker/Cura/files/6134765/cura.zip)

fvrmr, 2021-03-15:
Hi @GregValiant thank you for your bug report.
I am able to slice your model in Arachne beta. 
[image]

But the first time I tried, it didn't worked. Also @Ghostkeeper is able to reproduce this, so will we discuss this with the team. Keep you posted.

GregValiant, 2021-03-15:
Thank you.
It appears that when "Top/Bottom" was split from "Shell" that "Top/Bottom Pattern", "Bottom Pattern Initial Layer", and "Top/Bottom Line Directions" didn't get the memo.  FYI...I think that within the profile I used in that 3mf file, I had altered those settings from the defaults.

fvrmr, 2021-03-16:
I have created a ticket on the backlog.
Developers see CURA-8098

Ghostkeeper, 2021-04-01:
> It appears that when "Top/Bottom" was split from "Shell" that "Top/Bottom Pattern", "Bottom Pattern Initial Layer", and "Top/Bottom Line Directions" didn't get the memo. FYI...I think that within the profile I used in that 3mf file, I had altered those settings from the defaults.

As far as I can see, these are all correctly in the Top/Bottom category :confused:

GregValiant, 2021-04-01:
[image]
I will re-install Arachne and see how it goes.

GregValiant, 2021-04-01:
After re-installing (same download as before) there was no change in the displayed settings.  I have them checked to be visible but they don't show up and as a result, can't be altered.

Ghostkeeper, 2021-04-01:
@jellespijker This might be something to test in your current ticket.

jellespijker, 2021-04-01:
@Ghostkeeper I thought about that, but that would require a version upgrade from version 4000018 to 4000017 with all the dangers if circular upgrades, once the next version comes out.

aft7, 2021-04-30:
Here is a simple model and 3mf that shows the "slicing hangs" problem.

I first noted that slicing would hang if I set top layers to anything other than 0.
Then noted that it worked at full scale, but failed when scaled to 50%.
Then found : If I set Wall Line Count to "2" it works, but a higher value hangs.  Setting top layers to 0 still slices successfully (but not really useful...)
The model has 2mm thick walls, and I was using 0.5mm wall line width.

It appears to hang once the wall line count would be thicker than the actual walls when interacting with top layers?

Also can confirm - "Top/Bottom Pattern" is missing although it is checked in Setting Visibility - screenshot attached.

[arachne-fail.zip](https://github.com/Ultimaker/Cura/files/6402516/arachne-fail.zip)
[image]

fvrmr, 2022-01-11:
The ticket on our backlog is closed since we can't reproduce this with Arachne second beta. Can you confirm this @GregValiant?

GregValiant, 2022-01-11:
Yes, slicing went as expected and the settings are visible and available.  It looks good in Beta II so I'll close this.  Thanks Fenne.
[image]

closed 2022-01-11 by GregValiant