Ultimaker/Cura #9397: Unable to edit bed temp in profile settings

Scissors95, 2021-03-13:
### Application Version

4.8

### Platform

Windows 10

### Printer

Anet A8, Printrbot Simple Metal

### Reproduction steps

1. changed bed temp from 95 to 70 on right side bar.
2. opened profile manager and clicked the currently selected profile
3. clicked update profile with current settings/overrides
4. restarted cura

### Actual results

profile bed temp stayed 95

### Expected results

profile bed temp changed to 70

fvrmr, 2021-03-15:
Hi @Scissors95 thank you for your bug report. 
I am not able to reproduce this, could you share a project file?

And I am wondering if you are using the start optimizer plugin, this might have effect on it.

Scissors95, 2021-03-20:
@fvrmr  I do not have the start optimizer plugin added to cura. I have also attached a project file The bed temp is currently overridden in the print settings.

[CFFFP_CR10_hose_and_cable_clip.zip](https://github.com/Ultimaker/Cura/files/6175280/CFFFP_CR10_hose_and_cable_clip.zip)

fvrmr, 2021-03-23:
Thank you for your project file. 
I am still not able to reproduce your issue.
The build plate temperature stays at 70 after I updated the profile and restarted Cura. Does it happen on other profiles or with other settings?

Ghostkeeper, 2021-03-24:
I tried it too, but indeed your project file already starts with the temperature at 70. If I reset it, it goes to 95. If I change it back and restart Cura, it stays at the 70 degrees that I filled in.

I'm interested to see if there's anything in the log files about this. Perhaps there was an issue saving the state?

Scissors95, 2021-03-24:
I think we might be thinking about something different. I'm talking about editing the quality profile itself. The overrides stay after a restart. It refuses to let me override the bed temp in the quality profile.(attached) This occurs on all of my custom profiles across all 2.5 of my machines(one of them isn't set up fully yet)
[0.2mm layer.zip](https://github.com/Ultimaker/Cura/files/6200805/0.2mm.layer.zip)

Scissors95, 2021-03-30:
If it helps I've had this issue since before I had a multiple extrusion machine and when I first tried to fix it I set the profile bed temp to 70 on both of my active printers before attempting to save and close. It still would not update the profile.

Ghostkeeper, 2021-04-01:
Developers, see issue CURA-8148.

Scissors95, 2024-11-20:
I have switched slicers. Cura causes issues with my printers.

GregValiant, 2024-11-20:
I'll go ahead and close this.  Thanks for getting back.

closed 2024-11-20 by GregValiant