Ultimaker/Cura #9401: Drag and drop Z seam

Brissmoto, 2021-03-14:
### Is your feature request related to a problem?

Yes, it's pretty hard to set the seam exactly where you want. Sometimes with the settings one part of the seam is positioned properly while another part is not. And then it's impossible to have it like you want.

### Describe the solution you'd like

Now that we can see where the seam is exactly, it would be nice to be able to just drag and drop the seam where we want. The ideal solution we should be able to move the complete seam, selected parts of it or individual layers. Thanks for the nice work, it's appreciated. 👍

### Describe alternatives you've considered

Prusa slicer seam painter works pretty good for that...

### Affected users and/or printers

Everyone. Because everyone has to deal with z seam.

fvrmr, 2021-03-15:
Hi @Brissmoto thank you for your feature request. 
I will discuss this with the team. Keep you posted.

Ghostkeeper, 2021-03-29:
This was discussed two weeks ago, and it was decided it's indeed something we'd like to improve on. We've put this on our backlog to refine.
Developers, see issue CURA-8119.

BasF0, 2021-04-14:
I also looked into this briefly, but quickly abandoned adding that.. haha..

Anyway, to add my vision. I thought it would be nice to be able to select the starts that are in view, then allow dragging - where the starts snap to the mouse nearest line beginning. (Maybe a modifier key that would allow the start to be placed anywhere in a line - splitting it ?) For any starts in layers below or above the dragged start also snap nearest to the dragged one.

I would think direct manipulation of the starts could be preferred over painting the area where they should be in, maybe.

Ghostkeeper, 2021-05-03:
Dragging the starts in individual parts would get tedious very quickly with prints of 1000 layers, or with prints of 15 parts and each having 100 layers.

fvrmr, 2021-06-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. Right now it is already possible to change the seam. That does make the value of this feature smaller. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-07-13 by github-actions[bot]