Ultimaker/Cura #9404: Toggle / Setting for Extruding a Lead-In line to the Bed Adhesion Area or the first line of the Model Itself

quarky42, 2021-03-14:
### Is your feature request related to a problem?

Getting that first printed line to stick to the bed properly can be very difficult depending on filament type/condition, nozzle purging, and many other things.   Getting that perfect Start Code block that purges the nozzle or prints a little line on the edge can be very problematic.   

Sometimes there is too much distance between the purge area and the very first printed line.   Sometimes the first printed line doesn't want to stick very well at first.  Once it prints enough lines, it starts to stick really well, but the very first part of the first line printed often is loose and gets caught on the nozzle.  When that happens, that blob / booger can grab lines that did print well and pull them up off the bed.   This causes a snowball effect ruining the first layer and the whole print job.

For some printers, I've gotten some really great start codes to work where the filament is purged at the edge of the bed, the nozzle scrapes the filament off at the edge, and then moves very fast to the first line print location.  Other printers / firmware implementations are much harder to deal with because the initial line doesn't get printed right away.  That little bit of delay can really affect how that first initial line sticks.

### Describe the solution you'd like

I would like to have a "toggle" option that would let me print a single line called a "Lead In".  Think of how a vinyl record has a lead-in track that the needle drops onto and this track guides the needle to the start of the music.   The 3d printer version of this doesn't need to be quite so complicated, but if I could insert a single line print of variable length that leads into the first actual line that the slicer currently produces, that line should be well stuck to the bed after a sufficiently long enough path.   

So, with this option enabled, I would expect a print command inserted shortly after the GCODE Start Block, wherever the nozzle is, the "Lead In" option would move the nozzle down to the initial layer height and start printing a single line all the way over to the very first actual print line.    The only issue I foresee that when this option is enabled, a path that is outside of the initial line area would need to be calculated so that the nozzle ends up where the first line of the first layer starts.  So it wouldn't necessarily be a straight line over to the starting position, but a line that is calculated to use a free area of the bed on the way over to the first actual printed line that the slicer currently calculates.

If I could have one associated text field, it would be nice to specify the length of the lead-in.  This would be the length of the line as measured from the starting point of the first line within the print job currently.   If you selected a 50mm lead-in, the slicer would cause that line to be printed as if it were attached to the first normal line of the print (build plate adhesion area or model...whatever that first line is from.)  The slicer would create a path that fits on the bed outside of the printed area that starts the user selected distance away.   Some people might want a 50mm lead in.  I might want a 100mm lead-in because I really want to purge the nozzle more.  This could vary depending on the filament, but I suspect that there is a happy medium that would work for most jobs.

Because this lead-in avoids any other areas where printing is going to occur on the bed, it won't matter if the lead-in isn't fully attached at first.  It won't get in the way of anything.

### Describe alternatives you've considered

I've considered adding my own gcode, but I don't want to spend the time figuring out the amount of filament extruded per travel distance when the slicer is already doing that for me on the first line that is printed.... I just want that first line that is printed extended (on a non-interference basis) such that wherever the nozzle starts from, there is a single line printed leading into the model.

### Affected users and/or printers

I think everyone could benefit from this because if you print a purge line on your way into the first actual printed line, that first printed line should adhere very well by the time the nozzle gets there.   The lead in line wouldn't be in the way of the print job.

Asterchades, 2021-03-15:
I'm not exactly sure what separates this from a Skirt or Brim in the long run. Obviously a Brim will contact the model at more points where a skirt won't contact it at all (ordinarily, anyway) but they basically do exactly the same thing - just not in a straight line. They already adapt to the model size as well.

In fact, this seems to be exactly the purpose of a Skirt in the first place, seeing as it offers no actual mechanical advantage to the print.

quarky42, 2021-03-15:
No.  This is nothing like a skirt or brim.  In fact what I'm requesting could be used regardless of whether you've got skirt, brim, or no bed adhesion enabled.

If the very first line printed (regardless of whether it is skirt, brim, or model itself) fails to stick, but after a short amount of further printing it does start to stick well, then when it gets back around to the area where the loose bit is, the will catch on the nozzle and ruin the adhesion that has been established.  The loose bit will snag and pull the areas up that had been adhering.

By printing a lead in, into the first printed line, the bed adhesion can be established and the part that may be loose won't interfere with anything because it is away from all the other lines.

It's like the lead in on a vinyl record.

I often don't need a brim or a skirt if I can get the first line to stick well.   This isn't an issue for PLA but it can be an issue for petg and some others.

I am specifically asking to have the first line start printing away from anything else so that if it didn't stick in the first 10 or 30mm it wouldn't matter because the loose bit is nowhere near the area where the nozzle is going around and around.

quarky42, 2021-03-15:
"I'm not exactly sure what separates this from a Skirt or Brim in the long run."

Actually, to put is simply, the  very nature of what I'm requesting is that this one single printed line BE separated from the model, or if skirt/brim/raft is enabled that this line be separated from that.

The very nature of this request is to have one single line BE separate leading into the first print area so that if that line is loose somewhere it doesn't end up ruining the bed adhesion.

fvrmr, 2021-03-16:
Hi @quarky42 thank you for your feature request.
An easy workaround for this could be adding a prime line in your start g-code. Because this is unfortunately not a thing we would add to Cura. 

Here is an example for the start gcode:
```G28
G1 Y-3 F500 ; Move out of print volume
G1 X60 E9 F500 ; start purge line
G1 X100 E12.5 F500 ; finish purge line```

I will close this issue, but feel free to reach out if you have more questions.

closed 2021-03-16 by fvrmr

quarky42, 2021-03-16:
A manual prime line isn't the same thing.. there is still a gap between where it  primes and where it prints.   Printing a lead in line based on a calculated path outside of the model and outside of the brim/skirt/raft would ensure that the very first line of the model or brim/skirt was perfect.

fvrmr, 2021-03-16:
But what is than the value of that line while you can use the brim/skirt?

quarky42, 2021-03-16:
I've explained that.

The first line of any print, including the brim or skirt, doesn't always stick.  This is very material dependent so if you only print with a couple materials you might not have problems with this and I do.

After the first half of the first loop is printed it starts to stick just fine, but the loose bit at the very beginning gets snagged by the nozzle on the second pass.

If I could have a single line leading into the print area, that line could be loose at the beginning, stuck by the time it gets to. The print area and as the loops start to form on the bed the little loose bit doesn't snag on the nozzle because it is away from the print area leading into it.

quarky42, 2021-03-16:
I have 5 different printers road include borosilicate glass bed, buildtak, pei sheets, and coated glass.  They all exhibit this behavior to some degree depending on which filament I'm using.   One of the worst offenders is a clear PETG of a particular brand, but the colored ones don't stick like that.

I'm using line widths that are 1.5x the nozzle size, slow speeds, and my initial layer height is calibrated to within 0.01 and 0.02mm of what the slicer is set to.  My flow rates and esteps are all calibrated.  I get very good print results, but the first layer is a pain in the butt to have to watch it all the time.   I've watched this happen so many times that I have a pretty good idea when it is going to happen and it all has to do with a second line being laid down next to a loose first line.

fvrmr, 2021-03-23:
> If I could have a single line leading into the print area, that line could be loose at the beginning, stuck by the time it gets to. The print area and as the loops start to form on the bed the little loose bit doesn't snag on the nozzle because it is away from the print area leading into it.

Then you might wanna look into the retraction settings and change this to a lower value.

quarky42, 2021-03-23:
There is no retraction on the first line going into the first layer.  It retracts at the end of the first layer.  It would only retract again if it goes across an existing line which it doesn't.

If I'm missing something, please explain, because that goes against what I understand of how it works.

I don't think you understand what I've described.   When using concentric bottom layer the line doesn't stick until the filament has been coming out for a bit.   If I can have it star a line away from the model going towards the model it would be stuck great but the time it gets there.  Because it isn't stuck, the second loop starts and when it hits the loose section it snags and ruins the print.

This has nothing to do with retractions.

Calibration cubes come out with very flat lines on the first layer.  They are within 0.02mm of the Initial layer height.  The line width is 150% of the nozzle size.  I've tried 125% of the nozzle as well.  The issue is the type of filament I'm using.  It just doesn't stick very well on the first line until it has been printing for a short time.

A 50mm or 75mm line attached to the model as a lead in would do the trick. I don't have this problem with other kinds of filament, but this particular one just doesn't stick to the bed so well until it's been flowing.

I prime the nozzle before printing, but in the time it moves towards the bed it is isn't freshly primed.   If it was printing on the way to the first line, it would be nice and stuck by the time it got there.

Ghostkeeper, 2021-03-29:
We have a clash here because you're asking for a feature to prime the material before the structure is printed that is intended to prime the material. The skirt is intended to prime the material.

To make sure I understood correctly, let me try and summarise. The way that I read between the lines of your feature request, your problem is that the skirt doesn't stick to the bed because it's not primed yet when it starts printing that. You realise that your lead-in prime line would also not stick to the bed then, but you believe that wouldn't be a problem because the lead-in prime line is far enough away from the model that it doesn't interfere with the model any more even if it gets loose.

My counter argument is that the skirt is intended to be far enough away from the model that it doesn't interfere even if it gets loose. Perhaps it's just a matter of increasing the Skirt Distance? As it stands now, I would not see any advantage of printing this lead-in line over the skirt, and really just disadvantages since you wouldn't be able to adjust the amount of priming then.

Asterchades, 2021-03-29:
I feel like I've gotten a bit of an understanding what the request is about, so perhaps I can help.

When printing a skirt, particularly if the flow isn't quite right or the bed isn't perfectly clean (it happens to all of us at least once), the start won't adhere properly. There will be an extrusion, however, resulting in a piece of filament hanging clear of the print surface. This is mostly OK in the event the print head never passes near the area again until it's well clear vertically, but if it comes around for a second line of the skirt it could easily snag this loose bit, which then adheres to the nozzle, pulling up even more of the skirt material which can then get merged into the model proper. Having the skirt further away will simply mean more potential filament to pull up in the event this happens.

The proposed solution is to have an additional starting extrusion - or "tail", if you will - begin further away from the print, in a place where it won't come back into proximity of the nozzle again. That way if it fails to adhere correctly the nozzle (hopefully) won't pass over it again, and the risk of it being collected and subsequently deposited in the model is negated. The length of this tail would be configurable, such that the user could tune based on observations where the closest point of poor adhesion is going to be.

In effect it is functionally little different to a purge line, and indeed it could be used instead of one, but there is one key difference: there is constant, calculated, and deliberate extrusion from where it starts to the point where the next element begins (whether that's skirt, raft, brim, or the model).

Under ideal circumstances, I could see this being unnecessary. Ideally the whole skirt will stick, even if it doesn't necessarily start flowing immediately, and any deviation from that should be addressed in order to limit or prevent other issues (such as warping or outright delamination from the bed). If there weren't an automatic retraction between the start GCode and the print (see: https://github.com/Ultimaker/Cura/issues/9373) it may also be unnecessary. But things are rarely ideal in the world of consumer 3D printing.

quarky42, 2021-03-29:
Thank you, Asterchades, for taking the time to understand this request.  It is clear to me that you are the closest to understanding the request and the reason for it. The feature I am requesting is a calculated purge line called a lead-in purge line.  The differences from a gcode hard coded purge line are key.

A hard coded purge line is not connected to the model or to the skirt/brim.  The lead-in purge line would be connected to the model or skirt/brim (whatever is being printed first, the lead in would be placed leading up to it.).  The lead-in purge line would not be adjacent / touching any other line that is part of the skirt/brim or the model.  If the first part of the lead-in purge line didn't stick, being loose wouldn't matter because the nozzle isn't going to pass by that area again.  Some filaments like petg easily grab onto the nozzle more so than others.  Some are much more forgiving.

I 100% agree that if things are perfect, this request is unnecessary, but even on my freshly cleaned bed I have a type of translucent petg that simply will not stick at the beginning of the first loop of the model or the first loop of a skirt if that is enabled.  It sticks part way around the first loop but when the nozzle passes close to the beginning of the first loop on the start of the second loop, the loose bit is often caught by the nozzle and the whole thing is either ripped up or a blob attaches to the nozzle and ruins the print later on.   If I change filaments, I get varying results. Most stick very well.  I already use a purge before printing, and with most filaments that works very well, but it doesn't work for all filaments.  This translucent PETG and the very expensive NylonX both really have an issue sticking the first length of the first line printed but once they've been printing for a short time, they start to stick reasonably well.  NylonX is actually better behaved than this translucent PETG.

To be clear: my first layer is calibrated very well.  The lines lay down very flat next to each other, no valleys or ridges between the lines.  My first layer is within 0.02mm of what it is supposed to be.  I'm using thick 150% nozzle width lines.  The flow rate is dialed in.  This isn't my first time 3d printing.  It is because I have experience printing on 5 different machines that I see times where I would use this.  For this particular filament I would use this lead-in.  I have nothing else that I can calibrate for this filament to get it to stick reliably.  This feature would be immensely helpful in specific cases.  It is a smart purge line instead of a hard coded purge line on the edge of the bed where a long travel move is needed to get to the main print area.

quarky42, 2021-03-29:
Ghost keeper, there is no reading between the lines necessary.  I have clearly stated that I have observed the printed line adhere very well after a short time printing.  It is just the initial bit of the first line printed that doesn't stick well in some cases.  I have zero problems with most of the filament that I use.  There are a couple varieties where this lead-in would help a lot.

On a single loop of the skirt, it will often be adhering very well after 50 or 75mm of printing, but by the time the nozzle comes to the end of the first loop the loose bit gets snagged on the nozzle and rips the whole thing up.  If it doesn't do that, it often rips a part of the loose bit up, sticks to the nozzle, and that later gets snagged somewhere ripping the print off the bed.

Sometimes if I'm quick enough and if the first loop is large enough, I can clip the loose tail, but even cutting it pulls up a little bit that can get snagged.

I can hold the loose tail out of the way and the second pass will stick.  That loose tail being close to where the nozzle is can easily be picked up by the nozzle, though.

The difference here is you think I'm making a claim that can somehow be counter argued because you think all I have here is an untested theory.  I am actually making an observation after having printed with about ~30 different filaments across 300+ different prints.   3 of the filaments I use exhibit this behavior.  The other ~27 don't.   I have a work around that only works about half the time on those 3 problematic filaments because the work around isn't ideal, is done by hand, requires precise timing, and even then doesn't always work.  The work around still leaves a little bit if a loose printed line, but not as much.

Your attempt to argue this is itself theoretical based on what you think the problem is, where as I have hands on dealing with this problem numerous times and have watched this problem happen so many times with particular filaments, I can tell you with certainty that the root of the issue is the initial line being loose AND adjacent to another line being printed.  When I can eliminate that initial loose bit, the rest of the print is successful.  It's just really hard to do on a moving bed with a moving nozzle.

Ghostkeeper, 2021-03-29:
Okay, so to add to my summary: Your problem is that the start of the skirt line doesn't stick and that the nozzle then drags along the loose end when it comes close in a second loop of the skirt, if your skirt is small enough to need 2 loops.

This by itself is not really a problem though. The skirt is allowed to not stick. The nozzle is allowed to grab onto the skirt and drag it along. It really only becomes a problem if this skirt is then dragged along far enough to intersect with the placement model, where it could get stuck underneath the model. That is a problem because it results in an ugly line under the bottom of the end result, and a string that needs to be post-processed.

So what I responded was an alternative way to perhaps solve your issue. Increasing the skirt distance reduces the need to get a second skirt line (those 75mm are covered completely by a skirt distance of 12mm) and places the skirt farther away from the print so that even if it gets loose it's unlikely to move far enough to actually do any harm. Your feature is unlikely to get implemented, if such alternative solutions already exist. I'm just trying to help to solve your problem.

ton4eff, 2021-10-08:
Like this one? ;)
[image]