Ultimaker/Cura #9406: Printer is gone

czaras, 2021-03-15:
### Application Version

4.8.0

### Platform

Mac OS Catalina

### Printer

Ender 3 v2

### Reproduction steps

1. I was using Cura for I imported a profile which I've found on the internet. Import went smoothly and I could see all the options. Then I went to Properties to make some changes in General settings like currency and Cura crashed. After restarting all my Profiles for my Ender printer are gone and printer as well.

### Actual results

I've restarted Cura twice but every time the profiles didn't showed up.

### Expected results

I want my Profiles back.

no additional information
[cura.log](https://github.com/Ultimaker/Cura/files/6144966/cura.log)

fvrmr, 2021-03-16:
Hi @czaras thank you for your bug report. 
I am not able to reproduce this, but I would like to have the config files of the profiles. Because it looks like it has something to do with these. 
From the log files it looks like these are the files you have imported: 
Creality+Ender-3+v2.global.cfg, 
Creality+Ender-3+v2_user.inst.cfg, 
creality_base_extruder_0+%232_user.inst.cfg, 
creality_ender3pro_ender_3_v2_abs.inst.cfg, 
creality_base_extruder_0_ender_3_v2_abs.inst.cfg.

When I have these files I can try to reproduce this and see what is happening and why.
Thank you in advance.

czaras, 2021-03-16:
The profile I’ve imported is below:

fvrmr, 2021-03-16:
I can't see your profile because you've replied via e-mail.
Please add the profile (as .zip) in a comment on Github.

czaras, 2021-03-16:
[ender 3 v2 ABS.curaprofile.zip](https://github.com/Ultimaker/Cura/files/6149759/ender.3.v2.ABS.curaprofile.zip)

fvrmr, 2021-03-19:
Thank you @czaras I am still not able to reproduce this. I can import your profile and when I change some settings and restart Cura the printer and the profile are still there.
And don't see anything weird in log. So we will have to wait for others to have this issue to and hope that we can reproduce it the. 
Hope you understand.

czaras, 2021-03-19:
Ok, thank you for your support. maybe it was only one incidental case. 

have a good day!

fvrmr, 2021-03-19:
Yes that's possible. Let me know when you're having troubles again!

closed 2021-03-26 by fvrmr