Ultimaker/Cura #9409: Z Hop Not Functioning on Ultimaker 2+ profile

Airdoo, 2021-03-16:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Ultimaker 2+

### Reproduction steps

Have UM2+ selected and enable Z Hop When Retracted. 

### Actual results

Z Hops will not occur on machine nor appear in Preview tab as travels.

### Expected results

Perform Z Hops based on parameters selected.

fvrmr, 2021-03-16:
Hi @Airdoo thank you for your bug report. 
It is correct that you can't see z-hop in the preview tab. If you want could you share a project file (file - save project) and share it as a .zip here. Than I could take a closer look at it.

Airdoo, 2021-03-16:
Hi @fvrmr 

Here is the project and resulting gcode. Please let me know if there is anything else I can provide.

[UM2_zHop test.zip](https://github.com/Ultimaker/Cura/files/6151905/UM2_zHop.test.zip)

fvrmr, 2021-03-19:
Hi @Airdoo when you import the gcode in Cura then you see the zhops are there.
See here below.
[image]

So I am really curious what is happening at your side.

Airdoo, 2021-03-19:
Hi @fvrmr,

That's so odd, it's working now, I apologize for pulling dev time for this. Thank you for your assistance and keep up the great work

closed 2021-03-19 by Airdoo

fvrmr, 2021-03-19:
Hi @Airdoo no problem, good to know that it is working! No apologizes needed, it is my job after all 😄

Ghostkeeper, 2021-04-06:
See also https://github.com/Ultimaker/Cura/issues/5377.