Ultimaker/Cura #9411: [FR] Better rotation granularity

yet-another-average-joe, 2021-03-17:
### Is your feature request related to a problem?

The (displayed) rotation granularity is 1° if Snap Rotation is not checked ; I'm aware of #2779
We cannot know the exact rotation in free (No Snap) rotation mode.

### Describe the solution you'd like

Shift is used to toggle "Snap Rotation"
Ctrl and Alt are not used and could allow a better granularity, for example 1° and 0.5°, and even 0.1°
Ctrl -> 1°
Alt -> 0.5°
Ctrl + Alt -> 0.1°

Not as good as input fields, but the result would be the same (or nearly the same), at lower cost.

### Describe alternatives you've considered

Repositionning the model in CAD software.

### Affected users and/or printers

All users/printers

fvrmr, 2021-03-18:
Hi @yet-another-average-joe thank you for your feature request. 

Let me try to explain this, because this is not how the rotation is really working. It snaps to a pixel and that is not really 1 degree. So it be a bit complex to change this. 

Let me know if this makes sense :)

yet-another-average-joe, 2021-03-18:
Yes ! I think I understand what's explained in #2779

I was suggesting :

- keeping the snapping behaviour

- just changing the granularity (5°) for other values with dead keys : snap to 1, 0.5 or whatever, keeping the same mouse moves than those needed to change rotation in snapping mode

I don't know how rotation is coded. But I imagine that somewhere there's a function "xxx->rotate(angle)". In snapping mode, angle = 5° ; before calling the function, a test on dead keys could change the angle value for a smaller one.

Ghostkeeper, 2021-04-06:
The snapping of rotation is hard-coded here: https://github.com/Ultimaker/Uranium/blob/master/plugins/Tools/RotateTool/RotateTool.py#L49
This snap angle could be adjusted on the fly by a keypress, since it's evaluated only on mouse moves.

However honestly I think the use case for rotating in increments of 5 degrees is not all that useful. How often would you need to rotate by exactly 5 degrees, and not 5.1 or 4.9 degrees? I can imagine that it could be useful for models where you know that certain corners have 5 (or 95) degree angles and you want to align two surfaces exactly, but this is better and easier aligned in the 3D modelling software than in Cura. By far the most common case that's relevant for Cura is that somehow the coordinate system is messed up, rotating the model by 90 degrees. All other cases (e.g. support optimisation) don't require precise round-number angles or can be solved with the easier Lay Flat On Face function.

I would rather go for [this solution](https://github.com/Ultimaker/Cura/issues/2779) which would also serve those cases where a very precise rotation is necessary