Ultimaker/Cura #9412: Multiple strange behaviors  when using "Modify settings for overlaps" for infill.

pggh, 2021-03-17:
### Application Version

Arachne beta (printed) and 4.8.0 (according preview)

### Platform

Windows 10

### Printer

Anet A6 created from scratch

### Reproduction steps

For mechanical strength I wanted to have a some solid layers (instead of lower infill) in the middle of the part.
I created a support blocker cube, changed size and position and used it "Modify settings for overlaps" to change to 100% lines infill.

### Actual results

* The actual height (number of layers) where I got solid layers was much thicker than the height of the cube. It looks like that bottom and top layers are added. (I therefore reduced the height of the cube to 0.1mm to reduce the number of layers)
* 3 Layers in the middle are filled with lower speed than the ones below and above.
* The direction of these layers is not +-45° and in the preview the lines look thinner
* During printing, a discover that for these 3 layers, the cooling fan was been turned on, despite I disabled part cooling for this Nylon print.

### Expected results

Getting 100% infill only for the intersection of the volumes and no other side-effects.

[g1b.zip](https://github.com/Ultimaker/Cura/files/6155770/g1b.zip)

fvrmr, 2021-03-18:
Hi @pggh thank you for your bug report. 

I can't seem to import your whole file because your custom printer. 
But I can import your model (with cutting mesh)

> The actual height (number of layers) where I got solid layers was much thicker than the height of the cube. It looks like that bottom and top layers are added. (I therefore reduced the height of the cube to 0.1mm to reduce the number of layers)

This is happening because of the cutting mesh, so that's expected behaviour. 

> 3 Layers in the middle are filled with lower speed than the ones below and above.
> The direction of these layers is not +-45° and in the preview the lines look thinner
> During printing, a discover [...]

And these result are probably happening because of bridging settings see: https://github.com/Ultimaker/Cura/issues/9083

If you want to have solid layers in the middle part, you can use the infill mesh instead of cutting mesh. 
Please let me know if it works for you, and if you want that I take a look at it again, please share your project file.

pggh, 2021-03-20:
> If you want to have solid layers in the middle part, you can use the infill mesh instead of cutting mesh.
Please let me know if it works for you, and if you want that I take a look at it again, please share your project file.

Indeed, using infill mesh instead of cutting mesh solves the problem with creating bridges. I probably would not have made that mistake, if there would be a better documentation of the effect of this setting.

> During printing, a discover that for these 3 layers, the cooling fan was been turned on, despite I disabled part cooling for this Nylon print.

Originally caused by using "cutting mesh" I consider this a bug. When "Print Cooling" is disabled, the fan should never be turned on. If one would wish to only cool bridges, it would be logical to turn Print Cooling **ON** and use 0 for Regular/Initial Fan Speed and the desired Bridging Fan speed.

closed 2021-04-07 by Ghostkeeper