Ultimaker/Cura #9413: Crash with Accessibility "Hover Text" activated in macOS

gonzalezcalleja, 2021-03-17:
### Application Version

4.8.0

### Platform

macOS Catalina - intel i5

### Printer

Any printer

### Reproduction steps

Launch Ultimaker Cura, and then crash. 

### Actual results

Crash.

### Expected results

Not Crashing.

With this accessibility feature enabled (Hover Text on Mac) (more info  https://support.apple.com/en-gb/guide/mac-help/mchlb203bc78/mac), the application crash at start.

fvrmr, 2021-03-18:
Hi @gonzalezcalleja thank you for your bug report. 
Could you share your log file? You can find that here: `$USER/Library/Application Support/cura/<Cura version>/cura.log`

gonzalezcalleja, 2021-03-25:
Hi @fvrmr
here is the log file:

`
2021-03-25 21:06:11,014 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2021-03-25 21:06:11,025 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LegacyProfileReader
2021-03-25 21:06:11,035 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin AMFReader
2021-03-25 21:06:11,046 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SimpleView
2021-03-25 21:06:11,061 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SolidView
2021-03-25 21:06:11,076 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2021-03-25 21:06:11,087 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SentryLogger
2021-03-25 21:06:11,103 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SupportEraser
2021-03-25 21:06:11,123 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PrepareStage
2021-03-25 21:06:11,153 - WARNING - [MainThread] OctoPrintPlugin.OctoPrintOutputDevicePlugin.<module> [49]: Supplied version of Zeroconf module imported
2021-03-25 21:06:11,167 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OctoPrintPlugin
2021-03-25 21:06:11,186 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ImportExportProfiles
2021-03-25 21:06:11,198 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CustomSupports
2021-03-25 21:06:11,206 - DEBUG - [MainThread] UM.PluginRegistry.loadPlugins [432]: Loading all plugins took 1.5759971141815186 seconds
2021-03-25 21:06:11,225 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [163]: Looking for old configuration files to upgrade.
2021-03-25 21:06:11,271 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [175]: Checking and performing updates took 0.04462289810180664
2021-03-25 21:06:14,335 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [361]: Loading metadata into container registry took 2.7591991424560547 seconds
2021-03-25 21:06:14,364 - DEBUG - [MainThread] cura.Scene.CuraSceneController.setActiveBuildPlate [142]: Select build plate: 0
2021-03-25 21:06:14,369 - INFO - [MainThread] cura.CuraApplication.run [782]: Initializing machine error checker
2021-03-25 21:06:15,054 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition anet3d_a2_plus failed to pickle.
2021-03-25 21:06:15,079 - INFO - [MainThread] cura.UI.MachineActionManager.addDefaultMachineActions [74]: Default machine actions added for machine definition [anet3d_a2_plus]
2021-03-25 21:06:15,692 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition fdmprinter failed to pickle.
2021-03-25 21:06:15,809 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [194]: Extruder[anet3d_extruder_0 #3] added to [Anet A2 PLUS BUENA] at position [0]
2021-03-25 21:06:15,811 - INFO - [MainThread] FirmwareUpdateChecker.FirmwareUpdateChecker.checkFirmwareVersion [79]: No machine with name Anet A2 PLUS in list of firmware to check.
2021-03-25 21:06:16,239 - DEBUG - [MainThread] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for anet3d_a2_plus took 0.4074230194091797 seconds.
2021-03-25 21:06:16,244 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: ChangeAtZ
2021-03-25 21:06:16,246 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: ColorMix
2021-03-25 21:06:16,247 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: DisplayFilenameAndLayerOnLCD
2021-03-25 21:06:16,248 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: DisplayProgressOnLCD
2021-03-25 21:06:16,250 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: FilamentChange
2021-03-25 21:06:16,251 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: InsertAtLayerChange
2021-03-25 21:06:16,252 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: PauseAtHeight
2021-03-25 21:06:16,253 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: RetractContinue
2021-03-25 21:06:16,254 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: SearchAndReplace
2021-03-25 21:06:16,255 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: Stretch
2021-03-25 21:06:16,257 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: TimeLapse
2021-03-25 21:06:16,259 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: UsePreviousProbeMeasurements
2021-03-25 21:06:16,327 - INFO - [MainThread] cura.CuraApplication.run [787]: Initializing machine manager
2021-03-25 21:06:16,335 - INFO - [MainThread] cura.CuraApplication.run [792]: Initializing container manager
2021-03-25 21:06:16,521 - WARNING - [MainThread] UM.Logger.warning [128]: Starting a new sync while previous sync was not completed
{'CloudOutputDeviceManager': 0}
[... cut ...]
2021-03-25 21:06:21,414 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2021-03-25 21:06:21,423 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [62]: Updating QualityProfilesDropDownMenuModel.
2021-03-25 21:06:23,141 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 5.9s
2021-03-25 21:06:33,205 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles

`

Ghostkeeper, 2021-04-07:
From the log, I can tell that it's not crashing in Cura's code itself but in a C++ library that Cura uses. It is not logging anything about a crash, which means that the Python environment itself closed down, which must've come from an interrupt, such as a segfault. Going by your description, a segfault in the Qt library seems likely.

In Cura 4.9 we updated the Qt library to a more recent version (one that actually supports Catalina at all). So this might fix the issue for you. Would you like to try the Cura 4.9 beta that we recently released? https://github.com/Ultimaker/Cura/releases/tag/4.9-beta

gonzalezcalleja, 2021-04-07:
Hi @Ghostkeeper, thanks for your response.

I have tested right now with last 4.9 beta and now Cura doesn't crash but the "Hover text" accessibility feature doesn't crash, probably because this feature isn't implemented in QT library.

Thanks for your support!

Ghostkeeper, 2021-04-08:
That's possible, indeed. It's a fairly new feature of only one specific platform. I can't find anything about it in Qt's documentation, nor even in the [issue tracker of Qt](https://bugreports.qt.io/browse/QTBUG-86880?jql=text%20~%20%22hover%20text%20accessibility%22) so I think it's not even on their radar.

Cura should scale with the font size of your system, so one workaround would be to increase the font size there. But of course that makes *everything* larger so it's not nice to work with either.

I will close this, because without Qt supporting it there is nothing we can really do to fix this.

closed 2021-04-08 by Ghostkeeper