Ultimaker/Cura #9414: Uh when I preview this happens

TheSamCo, 2021-03-18:
*Archane Engine BETA**
(The version of the application this issue occurs with.)

**MacOS**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Ender 3**
(Which printer was selected in Cura?)

**clicked preview**

**Screenshot(s)
[image]
**
(Image showing the problem, perhaps before/after images.) 

**Went all red and weird and flashy**
(What happens after the above steps have been followed.)

**Show me the Preview for the model**
(What should happen after the above steps have been followed.)

**I cannot cos it doesn't work anymore sorry.**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file** ???
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-03-18:
Hi @TheSamCo thank you for your bug report. 
Does this only happen on the Beta of Arachne or also on 4.8?
And what MacOS are you using?

TheSamCo, 2021-03-22:
macOS Big Sur, and it only seems to happen on Arachne

Ghostkeeper, 2021-04-07:
The Qt version we were using at the time didn't support Big Sur.

With Cura 4.9 we upgraded to a Qt version that fixed a lot of things for Big Sur. We've observed that the other Big Sur issues have been fixed in Cura 4.9. We couldn't reproduce this issue to test if this issue is fixed too. Perhaps you'd like to give the 4.9 beta a try? https://github.com/Ultimaker/Cura/releases/tag/4.9-beta

closed 2024-12-08 by github-actions[bot]