Ultimaker/Cura #9416: very weird slicing

IxyR4, 2021-03-18:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Sliced model with settings for PETG

### Actual results

VERY weird layer making, see files.

### Expected results

I was expecting a normal slice.

[image]
[image]
[image]
[image]

SCREENSNAPS

CURAPROFILE
[https://drive.google.com/file/d/10tLTnR1bQ4zuh3NnkHA9MocZa_dQRTJs/view?usp=sharing](url)

**Thank you for help
//IxyR4**

IxyR4, 2021-03-18:
And it slices fine with my PLA settings

fvrmr, 2021-03-18:
Hi @IxyR4 thank you for your bug report. 
Could you share a project file (file - save project) instead of a cura profile and share it in a comment on Github as a .zip.
So I can take a look at it.

Thank you in advance.

IxyR4, 2021-03-18:
Yeah sure, just give me some hours and i'll be home

closed 2021-04-02 by no-response[bot]

Ghostkeeper, 2021-04-08:
I think the issue is simply that the parts are too thin to print. But there's no telling without having access to the project file.