Ultimaker/Cura #9418: ChangeAtZ 5.2.1

see3d, 2021-03-18:
### Application Version

4.8.0

### Platform

Mac Mini M1 Big Sur

### Printer

MP Delta Pro

### Reproduction steps

Put in a temp change for all layers on and after layer 3.  I had 5 duplicate parts printed one at a time.  

I tried to change at height instead.  Height does not seem to be a workaround either.

### Actual results

It changed the temp on the first part, then did not for the next 4 parts.  Layer number does not start over for each part as far as ChangeAtZ goes.  

 I tried to change at height instead.  That did not work at all (I did have Z Hop on retract enabled which is an issue from the other old bugs).  

### Expected results

Temp should change at height on each part printed, even if the layer number does not start over.

Ghostkeeper, 2021-04-08:
Looks like a duplicate of https://github.com/Ultimaker/Cura/issues/8555 to me.

see3d, 2021-04-09:
Sorry for no response.  I was having trouble pinpointing where the error was.  I ran into a case where the gcode that was being generated even without the ChangeAtZ active was not setting the temperatures right for the initial layer (setting it immediately then resetting it a few gcode lines later).  This may be the actual source of the bug.  I will have to make a new issue if I can not locate another similar issue already reported.

Ghostkeeper, 2021-04-15:
I'm not quite sure what you mean. You say that a temperature change g-code is being generated even if the ChangeAtZ is not active? A temperature change not related to the printing temperatures or stand-by temperature?

closed 2021-04-29 by no-response[bot]