Ultimaker/Cura #9419: Cura doesn't start with active Autodesk Application manager

digiTALker, 2021-03-19:
### Application Version

4.8.0

### Platform

Win 10

### Printer

UltiSteel

### Reproduction steps

Just try to launch Cura when Autodesk App Manager in tray.

### Actual results

Cura is blink in task manager and than disappear.

### Expected results

Expected to launched Cura

fvrmr, 2021-03-19:
Hi @digiTALker thank you for your bug report.
Could you share your log file? Then I can maybe see what is happening and why.
You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

digiTALker, 2021-03-19:
> cura.log

[cura.log](https://github.com/Ultimaker/Cura/files/6172987/cura.log)

also after Cura started possible to run App Manager and both progs will works well.

fvrmr, 2021-03-26:
Nice to hear that it is working now. I will close it, but reach out if it happens more.

digiTALker, 2021-03-31:
Its working only if I closed app manager

Ghostkeeper, 2021-04-08:
I wouldn't know what the Autodesk App Manager does that could interfere with Cura. Maybe you have the Fusion 360 integration plug-in installed from the Marketplace and that is somehow crashing Cura when it talks to Fusion 360 or something?

The log file says basically nothing relevant. It shows that you successfully sliced some things (slicing took 7-10 seconds every time) and were looking at layer view. So something must've worked. Maybe that was a log from when the Autodesk App Manager was closed.

digiTALker, 2021-04-27:
> I wouldn't know what the Autodesk App Manager does that could interfere with Cura. Maybe you have the Fusion 360 integration plug-in installed from the Marketplace and that is somehow crashing Cura when it talks to Fusion 360 or something?
> 
Yes I have this plugin but didn't used it for a long time.

> The log file says basically nothing relevant. It shows that you successfully sliced some things (slicing took 7-10 seconds every time) and were looking at layer view. So something must've worked. Maybe that was a log from when the Autodesk App Manager was closed.
You can't see any new info in logs cause Cura can't start. And you see only logs from prev correct work.

Ghostkeeper, 2021-05-13:
Ah, so then it's not just the Fusion 360 integration plug-in that causes the issue. The code for that plug-in hasn't even been loaded yet, because by that time the logger would already have said something. It seems that this Autodesk App Manager is preventing Cura from executing any code whatsoever, or at least something very close to it.

I'm not sure what we could do to prevent this? I can't write any code that fixes it if my code doesn't get executed, you see.

But I'm still not sure why Autodesk App Manager would do anything like that.

Ghostkeeper, 2021-05-13:
The log file also doesn't show you opening the application successfully, nor closing it. It appears like you just cut a piece of the log from the middle somewhere from a time when Cura was running successfully.

Perhaps you can share the whole log file?

closed 2021-05-27 by no-response[bot]