Ultimaker/Cura #9420: brim doesn't add to second item if any part is rotated

jxer, 2021-03-19:
### Application Version

4.8

### Platform

windows 10

### Printer

ender v2

### Reproduction steps

added a nut and a screw, when spinning either piece the screw will not add any form of support, remove the nut object and brim comes back

### Actual results

only 1 item receives a brim

### Expected results

both should receive brim

[screw.zip](https://github.com/Ultimaker/Cura/files/6173758/screw.zip)

attached

jxer, 2021-03-19:
correction, simply adding the screw and rotating 180 degree's stop the item from creating a brim

fvrmr, 2021-03-22:
Hi @jxer thank you for your bug report.
Could you add a project file (file - save project) instead of stl files.
Because when I import these two models both do have a brim, but that is with my settings. A project file contains your settings and printer. So it would be easier to reproduce your issue.
[image]
Thank you in advance.

jxer, 2021-03-22:
spin the screw 180  degree's please so the head is resting at the bottom

fvrmr, 2021-03-22:
Ah thanks, but get still a brim. 
[image]

If you want me to look at it again a project file is very helpful.

jxer, 2021-03-22:
[CE3PRO_nut.zip](https://github.com/Ultimaker/Cura/files/6183691/CE3PRO_nut.zip)
[image]

not doing anything special, rotate both items and boom i can't get the screw to take a brim, nor does it print one either.

fvrmr, 2021-03-23:
Thank you!
The reason why there is no brim created is because the screw does not touch the build plate.

See here at layer 1:
[image]

By dragging in to the build plate 0.09 mm the screw touches the build plate and the brim is created. 
[image]

But it is weird that I get a brim without dragging the screw into the build plate, switching to different profiles it seems like that's the issue. Could you check if this is also the case for you?

jxer, 2021-03-25:
do you mean quality profile? shouldn't it be snapping to the build plate? it does when its a single piece. definitely seems like a bug

Asterchades, 2021-03-25:
Check the Mesh Fixes section, making sure that Remove Empty First Layers is checked. That's the only profile-level setting I can think of which might allow that to occur.

GregValiant, 2021-03-25:
The screw is touching the build plate with the Z = 0.  Because it's an oval head phillips screw the points of the screwdriver cross that are touching the build plate are so miniscule (borderline "infinitely small") that the contact patch is effectively zero so Cura is ignoring them either because they can't be printed, or their size falls below how ever many decimal places Cura works to.  I think that's why there isn't any brim since for that part in that Z location there is no first layer.  I was able to generate a brim with the Z at -0.01 but there is still no part.  AT Z = -0.02 the part starts to print on layer #1.  Unfortunately, because it is an oval head the rest of the head doesn't touch the build plate and would require support.  Changing the part to a conventional flat head screw makes printing much easier with the head down and I'm sure it would look better.  It should be possible to print the oval head with the head "up".

jxer, 2021-03-25:
this issue only occurs if there are 2 pieces. I don't have to play around with z axis, if the screw is by itself, only when the nut is added.

GregValiant, 2021-03-25:
If I move the nut off the build surface then yes, the screw has a brim at .12 Initial Layer Height so something is going on there between the parts.  Grouping them together had no effect.  With the Initial Layer Height at .16 and both parts on the surface they both have a brim.
I'm not giving up yet on my "Infinitesimal Points" theory just yet.  I'm pinning my hopes on the fact that a taller 1st layer height fixes the problem.

fvrmr, 2021-03-26:
> do you mean quality profile? shouldn't it be snapping to the build plate? it does when its a single piece. definitely seems like a bug

Yes with super quality 0.12 mm the brim is gone and changing it to dynamic quality 0.16mm the brim is added.

> I'm pinning my hopes on the fact that a taller 1st layer height fixes the problem.

Yes changing the initial layer height to 0.16mm is a workaround.

jxer, 2021-03-26:
is this going to be listed as a bug and resolved?

fvrmr, 2021-03-26:
I will discuss it with the team. Keep you posted!

fvrmr, 2021-03-30:
Hi @jxer I have discussed it with the team. But we don't see this as a bug, so we won't fix this. 
The reason why we don't see this as a bug has to do with the models. Because the nut is flat and the screw head not, the initial layer of the screw is less than 0,12 mm so there is no layer. That explains why it is visible with another profile. 
Hope this makes sense to you. Please reach out if you have more questions.

Ghostkeeper, 2021-04-08:
> this issue only occurs if there are 2 pieces. I don't have to play around with z axis, if the screw is by itself, only when the nut is added.

This is because of the Remove Empty First Layers setting. If the screw is the only item on the build plate, the first layer is empty and it then gets removed, effectively moving down the screw by 0.1mm. However if you also have a bolt on the build plate, the first layer is no longer empty. Cura assumes that the screw is supposed to be in that position relative to the bolt, and so it won't automatically move the print down any more.

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant