Ultimaker/Cura #9422: Cura breaks on MacBook Air (M1)

Pheerink, 2021-03-20:
### Application Version

Cura 4.8

### Platform

MacBook Air (MacOs Big Sur)

### Printer

Artillery X1 V4

### Reproduction steps

Slicing Works fine until you do a preview. Then the program hangs.

### Actual results

Distortion of the screen and program hangs

### Expected results

Normal operation

fvrmr, 2021-03-22:
Hi @Pheerink thank you for your bug report.
This is a duplicate of #8745.
We are working on a solution and we already have done a test build that will fix this problem. Progress with the upgrade of the Qt version is still ongoing. There are other crashes still to solve now.

Because this is a duplicate I will close this issue.

closed 2021-03-22 by fvrmr

Pheerink, 2021-03-22:
Sorry couldn’t find it. Thanks anyway

Regards
Paul