Ultimaker/Cura #9423: Slicing issue with Leapfrog Bolt Pro

JimF42, 2021-03-20:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Leapfrog Bolt Pro

### Reproduction steps

Installed a new instance of Cura 4.8.0. 
Added the built-in Leapfrog Bolt Pro profile.
Added a model to the bed (doesn't appear to matter which one).
Set the filament to eSUN White PLA+ (doesn't seem to matter).
Changed Infill Density and/or Infill Pattern.

I have tried this on both my desktop computer and laptop computer with the same results.

### Actual results

The Infill Density AND Infill Pattern are being ignored. The sliced model, when previewed, will always have a GRID pattern and the infill density setting does not change in the sliced model either.

The attached project shows the results, along with a screenshot of the preview.

### Expected results

I expect the sliced model to honor the requested Infill settings.

There may be other settings that are being ignored, but I am not aware of them as I don't usually tweak other settings.

[Cura Slicing Issue.zip](https://github.com/Ultimaker/Cura/files/6175573/Cura.Slicing.Issue.zip)

GregValiant, 2021-03-21:
With the right extruder disabled it works.  Once.
Switching back and forth between extruders and disabling/enabling extruders gives conflicting results.  Sometimes the infill changes and sometimes it doesn't.

JimF42, 2021-03-21:
I'm still thinking it's a bug, but, I think I got it.

I had to **unhide** and **explicitly set** the Infill Extruder value. By default, it is set to "Not overridden". When I set the Infill Extruder to be the same as the rest of the model, it now seems to work.

I'm still thinking it's a bug and at the very least, when there are multiple extruders, _maybe_ Infill Extruder shouldn't be hidden. But, I would think if the Infill Extruder is left as "Not overridden", then the infill parameters should still apply to all extruders.

[image]

fvrmr, 2021-03-22:
Hi @JimF42 thank you for your bug report.
This is happening because you have changed the settings in Extruder 1 instead of extruder 2. 

Extruder 1:
[image]

Extruder 2:
[image]

The solution for this is changing the settings for the correct extruder you are using. I know it is a bit confusing.
Let me know if you have more questions!

JimF42, 2021-03-22:
Arrggg. Yup, that's what I was doing.

So, part of this confusion is that the RIGHT extruder is #1 and the LEFT extruder is #2 and so the Right extruder shows up on the Left tab and the Left extruder shows up on the Right tab. :man_facepalming: It's the same issue with model/extruder selection too.

Is there something I can do in my copy of this profile to switch them around and still have it work correctly?

fvrmr, 2021-03-26:
Hi @JimF42 that's confusion indeed but you can't change the interface easily. Maybe you can change it in your printer settings, that the number above the extruders are the same like the settings. But I am not sure.

closed 2021-03-31 by JimF42

Ghostkeeper, 2021-04-08:
You can right-click on a setting and select "Copy value to all extruders" to bring it over to the other extruder. Or "Copy all changed values to all extruders" to move all user-changed settings at once.