Ultimaker/Cura #9427: Using ChangeatZ 5.2.1 with Speed/Print Speed changes, elapsed time is not updated in the GCode or the summary tab in Cura

arbie666, 2021-03-22:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Load any model, i.e. part calibration cube
ChangeatZ 5.2.1 Set change at layer 1 and select either or both speed or print speed.
Set speed to a different value, i.e. 200% and reslice.
No change to elapsed_time in GCode notes or on the Cura information tab.

### Actual results

Time unchanged

### Expected results

In this example time should be virtually halved

fvrmr, 2021-03-22:
Hi @arbie666 thank you for your bug report. 
It is correct that this happening, let me explain why.
When you press slice the post processing script are not calculated in the estimated time. This happens after you press save. 
And the reason why this is not working is because it is very complex to implement this, that's why I will defer this issue.
Hope you understand!

arbie666, 2021-03-23:
Thanks for the response.

Massive thank you to all that have worked on this program. It's amazing!
I've dabbled in programming so know nothing is ever simple.  A shame about
the time, but important thing is it changes the print code correctly, which
it does!  Although a seasoned engineer, I'm a newbie to gcode and been 3d
printing only a few months thanks to furlough.  There's so much to learn
and play with.  Every day is a learning day.

Keep safe, regards, Rich.

closed 2023-06-13 by github-actions[bot]