Ultimaker/Cura #9428: Include power consumption in cost calculator

samy, 2021-03-22:
### Is your feature request related to a problem?

Not a problem

### Describe the solution you'd like

A field to add power cost per hour 

### Describe alternatives you've considered

Calculate by hand :)

### Affected users and/or printers

All users with cost estimation displayed

It should be useful to include basic power consumption cost in cost estimation (to have an idea of the cost). Not mandatory to have very precise system, with night power cost, etc

StuSerious, 2021-03-22:
Dupe? More discussion here: #9261

fvrmr, 2021-03-22:
Hi @samy thank you for your feature request.
Like @StuSerious mentioned this is a duplicate of #9261
That's why I will close this issue. Feel free to continue the thread in the other issue. 
Hope you understand.

closed 2021-03-22 by fvrmr