Ultimaker/Cura #9429: Pausing doesn't work.

jeq1, 2021-03-22:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Geeetech A30Pro

### Reproduction steps

1. Tried every pause script and filament change.
2. Furstated.
3. Took beer.
4. Contacted seller of 3d printer, he told that every printer has same problem on Cura right now.

### Actual results

M0 leads nozze to raise, then go back down and continues printing.
M25 leads for inifite loop of pause, when user presses play, it seems like its going to print but comes back very short after, before printing anything.

### Expected results

M0 should pause printer so user have to press play.
M25 should pause printer so user have to press play.

GregValiant, 2021-03-22:
This is probably the same firmware problem that the Ender's were having.  The printer doesn't communicate with the display correctly and bypasses the command.  M0, M1, and M117 were the problems on the Creality machines.  On the Creality machines it is directly related to the TFT type of screen.
One thing that kind of works is to use the M0 command and add "G4 S180" in the "Gcode after Pause" box.  That gives you a 3 minute window to change filament and then the printer will restart.  You can make the dwell longer as "S" is in minutes.
If you use the M25 command - is "Pause" one of the options on the LCD after the printer moves to the park position?  If it is then you may be able to "Pause" on the LCD and immediately "Resume".

The Marlin command M0 means "Unconditional Stop".  The Marlin command M25 means "Pause SD Print".  The commands are in the Gcode file so it is up to the firmware to accept the command and do what it is supposed to do.

fvrmr, 2021-03-23:
Hi @jeq1 thank you for your report.
Like @GregValiant mentioned this is also a firmware problem. Could you try if his suggestion works for you?

closed 2021-04-06 by no-response[bot]