Ultimaker/Cura #9430: Pause At Height (Post Processing 2.2.1) returns nozzle to incorrect position

pnt103, 2021-03-22:
### Application Version

4.8.0

### Platform

Windows 10 Pro 64-bit

### Printer

Ender 3

### Reproduction steps

Used PauseAtHeight to insert pause at a particular layer (rather than a particular height) to facilitate filament change.

### Actual results

If the layer is the penultimate layer, the nozzle returns to 0,0 afterthe  pause, instead of the correct X,Y position to restart.  If layer is lower than the penultimate, the nozzle moves to a position guessed from the first extrusion two layers ahead, which is still incorrect and may or may not be within the model.

### Expected results

Nozzle should return to correct X,Y position to start the next layer.

I discovered two problems with the plugin, which does not return the nozzle to the correct position after the pause.

I've traced the problem to the following 4 lines in the PauseAtHeight.py script, which are the cause of the problem:

    # Get X and Y from the next layer (better position for
    # the nozzle)
    next_layer = data[index + 1]
    x, y, myxyline = self.getNextXY(next_layer)

Firstly, _data[index+1]_ is not the next layer to be printed, but the layer after that. The next layer, ie immediately after the pause, is _data[index]_.  If the PauseAtHeight is between the penultimate layer and the final layer, the existing code will try to get values beyond the end of the file, usually resulting in x,y = 0,0 because _getValue()_ returns _None_ on error.

Secondly, what's actually required is the start position of the next layer, which is the final X,Y pair from the travel moves in the NONMESH section after the layer just completed.  The above code would get the X,Y position of the first command in the indicated layer, which would instead be the _end_ of the first _extrusion_ in the indicated layer.

I've attached a script with corrected code which I've tested with several models, and an STL file (skewed block v1.stl) I used for testing.  Note that I have not edited the section starting "# Maybe redo the last layer", which presumably is also incorrect.  The test STL file was constructed in such a way that I could test that the script was reading the X and Y values from the correct layer, because every layer starts and ends in a different but determinable position from every other.  I've also included examples of project files made with the corrected script.
[script_and_tests.zip](https://github.com/Ultimaker/Cura/files/6185956/script_and_tests.zip)

pnt103, 2021-03-23:
To see what I mean about the problems without having to install my script, I've attached three small gcode files in the attached zipfile, one made with my "fixed" script, two with the original to show the two aspects of the problems.

These three gcode files were created by slicing my "skewed block v1.stl" file, which was lowered below the build surface to make it smaller: only 5 layers in total.

skewed block v1 fixed.gcode:
- Created with my "fixed" PauseAtHeight.py script.
- The pause is between the penultimate and final layers.
- The nozzle moves to the correct position after the pause.

skewed block v1 X0Y0.gcode:
- Created with the original PauseAtHeight.py script from Cura 4.8.0
and Post Processing 2.2.1
- The pause is between the penultimate and final layers.
- The nozzle moves to 0,0 after the pause and extrudes filament in thin
- air as it executes the first command in the succeeding layer.

skewed block v1 XwrongYwrong.gcode:
- Created with the original PauseAtHeight.py script from Cura 4.8.0 
and Post Processing Plugin 2.2.1
- The pause is between the anti-penultimate and penultimate layers.
- The nozzle moves to a position derived from the next-but-one layer
and extrudes filament inappropriately, missing the correct portion of
the inner wall.

[gcode from tests.zip](https://github.com/Ultimaker/Cura/files/6186057/gcode.from.tests.zip)

rburema, 2021-03-25:
Hi!

Thank you, this is a _very_ nice bug report. In fact, since a patch is included, we'll be treating this as a Pull Request internally. (For the other devs, please refer to internal ticket CURA-8116)

I have discussed this with our community manager and one other dev, reproduced the problem, and will (probably, barring interruptions or unexpected problems) apply this patch to our main branch within the next few hours or so.

rburema, 2021-03-25:
Thanks once again! I've made a commit here: https://github.com/Ultimaker/Cura/commit/2ed40fe2c6e72814cdcd9abbaaa2e2bb95c4c7a0

If QA, ST or the beta users won't find anything, this will be in the upcoming 4.9

closed 2021-03-25 by rburema

Scubadooper, 2022-06-18:
This bug seems to still be in 4.13.1

Pause at height (10.8, 0.3 draft) returns nozzel to 11.1 (layer 35) layer 36 remains at 11.1

pnt103, 2022-06-20:
My fix seems to only affect pausing by layer number.  It doesn't necessarily work when pausing at a specific height rather than a specific layer.

commit 2ed40fe2 (2021-03-25): Get last position from previous layer instead.

Old code was next-next layer, which is actually the layer _after_ the one that is meant.

Fixes #9430
CURA-8116