Ultimaker/Cura #9432: Make changes from the selected profile always visible

dwery, 2021-03-23:
### Is your feature request related to a problem?

I change a setting that is not currently visible and there is no indication in the UI

### Describe the solution you'd like

Have an option to make settings that are changes from the currently selected profile always visible

### Describe alternatives you've considered

n/a

### Affected users and/or printers

All

fvrmr, 2021-03-23:
Hi @dwery thank you for your feature request. 
If you will click on recommend you will see an arrow that will show changes you have made. 

For example:
I have my settings visibility on basic and I change the `line width` from 0.35mm to 0.8mm by searching for line.
[image]

Then I go to recommend 

[image]

And you will see a arrow and if you hover over the arrow this message comes up:

[image]

Please let me know if this works for you.

dwery, 2021-03-23:
Hi, this helps noticing it, but in order to know what changed you would still have to go back to custom, expand everything and check for the mark. It would be much easier to just make changed settings always visible.

yet-another-average-joe, 2021-03-26:
Aren't recommended settings for Ultimaker printers only, with listed filaments ?

This being said, an option in the filter menu ("Modified settings", relative to the current profile) would be very usefull.

Ghostkeeper, 2021-04-09:
> Aren't recommended settings for Ultimaker printers only, with listed filaments ?

No, the "recommended" settings are simply the default profiles without any individual modifications. Every dot is one of the profiles that would be available from the quality profile drop-down. These profiles are created by contributors in most cases (although some printers use the default "global" set of profiles, like Custom FFF Printer, which was originally based on the Ultimaker 2 profiles).

We've had an issue in our backlog for a long time to implement a setting visibility pre-set option for "all currently changed settings". Developers, refer to issue CURA-5421. Our product managers also seem to find this useful (and I agree too) but while it's on our priority list, it's still about halfway down the list right now.