Ultimaker/Cura #9435: Cura will not open after install

jdandy46, 2021-03-23:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

I recently Purchased a new HP laptop that runs on Windows (the newest version I assume). I am able to download Cura 4.8 from the Cura website. After downloading, Cura will open and I can add my printer and manipulate files, however, once I close and attempt to open Cura again, it prompts me to uninstall the "old version" before installing the new version (which I just installed). This happens every time. Is there a way to fix this? 

### Actual results

The program runs in administrator mode. when I close the program after initial install, it prompts me to un install.

### Expected results

II do not want to have to do the above steps. I just want the program to work correctly.

drsurfer, 2021-03-24:
Are you really sure you're clicking the program Icon in start menu, and not the installation program again?
Try to go in C:\Program Files\Ultimaker Cura 4.8.0 and double click on Cura icon there.

fvrmr, 2021-03-25:
Hi @jdandy46 thank you for your bug report. 
Like @drsurfer mentioned above, could you please check if that works for you. 
Let me know if this works for you.

jdandy46, 2021-03-25:
So the program is working now. I believe it was installed incorrectly. Thanks

fvrmr, 2021-03-26:
Great to hear. I will close this issue since it is working!

closed 2021-03-26 by fvrmr