Ultimaker/Cura #9436: Print via cloud

Thes3D, 2021-03-23:
### Application Version

4.6.3 Enterprise

### Platform

Windows 10

### Printer

Ultimaker 2+ Connect

### Reproduction steps

Setup UMC on network
Connect printer via digital factory
Digital factory sees printer but can not connect because it says it is in different group

### Actual results

Can not print via cloud

### Expected results

Print via cloud

This may seem minor, but clients are paying extra for this feature and it does not work on Cura Enterprise 4.6.3
Is there a more recent version than 4.6.3 for Enterprise?

fvrmr, 2021-03-25:
Hi @Thes3D thank you for your bug report. 
Could you try to add your printer manually in Cura Enterprise via ip adress? And see if that works.

And 4.6.3 is the most recent version for Enterprise.

Thes3D, 2021-03-27:
Hi,

Yes I tried with the IP address for the printer and it doesn't work either
manually.

fvrmr, 2021-03-30:
Could you share your log file in a comment on Github?
You can find you log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

fvrmr, 2021-03-30:
Hi @Thes3D please ignore my previous comment because you can't connect your UM2+ Connect in Cura 4.6.3 Enterprise because the UM2+ C profile wasn't in the release of Cura 4.6.3.

Ghostkeeper, 2021-04-09:
This is supposed to be fixed with release 4.9. Or rather, the UM2+C printer was added in 4.7 but the next Enterprise release is supposed to become 4.9, so it will be available then, once that is stable.

Thes3D, 2021-04-09:
Ok, thanks for the update.

Kind Regards,

Jaime Cavazos, CEO
THES3D
3D Solutions Partner
Zaimi 43, Triandria
55337 Thessaloniki, GREECE
Tel. +30 2313 052 143
Fax. +30 2313 052 144
Site. thes3d.gr

Subscribe to our Newsletter here
<https://thes3d.us10.list-manage.com/subscribe?u=76c7d3734086dfb36f2b77554&id=8a071fc1e4>

closed 2024-12-08 by github-actions[bot]