Ultimaker/Cura #9438: Custom support plugin

f350in1208, 2021-03-24:
### Application Version

4.8.0

### Platform

windows 10

### Printer

ender 5 plus

### Reproduction steps

1. attempted to save project to computer using all possible formats

2. opened saved files in Cura individually 

### Actual results

either files wouldn't load or files load but with all supports as the actual end project

### Expected results

custom supports to be saved properly so that they may be changed or updated without starting over after save

[image]
[image]

fieldOfView, 2021-03-24:
Please also post a link to the project you created. It looks like you did not save the project using File -> Save Project..., but instead exported the meshes using File -> Export... and used as the file format. That latter does not save a project but meshes only. If you want the project to include settings, you *must* use File -> Save Project...

f350in1208, 2021-03-24:
you're correct i was exporting instead of saving. thank you for your corrections and your time and sorry for the inconvenience.

closed 2021-04-08 by no-response[bot]