Ultimaker/Cura #9441: 4.8 ender 3 and started layer shifting.

realacc, 2021-03-24:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

GregValiant, 2021-03-24:
Nice.  Did you check the belts?  They should twang like the strings on a bass guitar.

realacc, 2021-03-24:
Yup I did, it's definitely not mechanical. Its a ender 3 running cura 4.8. Weird thing is this started layer shifting right after a windows update. Tried 3 different filaments and tried removing cura and re installing it. Also removed the stl file. Guess I should try printing a different file to see what happens. I was reaching out though to see if anyone else was reporting later shift with 4.8.

GregValiant, 2021-03-25:
If you would fill out the form and upload the gcode file then tomorrow the Cura developers will take a look.  Kinda hard to figure out with no info.  You could also open the gcode file in Cura to see how it looks.

jellespijker, 2021-03-25:
Please fill in the form and attach project files. We can't help otherwise.

closed 2021-04-08 by no-response[bot]