Ultimaker/Cura #9443: Please also post a link to the project you created. It looks like you did not save the project using File -> Save Project..., but instead exported the meshes using File -> Export... and used as the file format. That latter does not save a project but meshes only. If you want the project to include settings, you *must* use File -> Save Project...

f350in1208, 2021-03-24:
Please also post a link to the project you created. It looks like you did not save the project using File -> Save Project..., but instead exported the meshes using File -> Export... and used as the file format. That latter does not save a project but meshes only. If you want the project to include settings, you *must* use File -> Save Project...

_Originally posted by @fieldOfView in https://github.com/Ultimaker/Cura/issues/9438#issuecomment-805557257_

f350in1208, 2021-03-24:
it did save properly this time unfortunately for some reason I'm unable to edit individual supports. it wants adjust the entire project. 
[image]
am i doing something wrong?

f350in1208, 2021-03-24:
https://github.com/f350in1208/saving-issue/blob/main/CE5P_updated%20zia%20golf%20course.3mf

fvrmr, 2021-03-25:
Hi @f350in1208 I can't access the link you have posted above. 
Could you add your project file as a zip in a comment in the other thread #9438. 
I will close this issue and reopen the other one, so we keep everything on one place.

closed 2021-03-25 by fvrmr

Ghostkeeper, 2021-04-09:
I think that repository is made private, so we just can't access it.