Ultimaker/Cura #9444: Moniter tab showing nothing. ender -3

bryceOctovioOstrem, 2021-03-24:
### Application Version

4.8.0

### Platform

windows 10

### Printer

ender -3

### Reproduction steps

plugged in printer with cable into pc
clicked on the monitor tab

### Actual results

there is no interface despite the printer being properly set up

### Expected results

an interface with options would appear for printing

pls fix this. printing via usb is a pain

tyindeepit, 2021-03-25:
Hi, _I connect to my printer via OctoPrint_ but the very first time I tried to use it I had to go in to preferences > printer and then click "connect" button first.

I have no idea if direct USB connection would require you to click connect or not though.

GregValiant, 2021-03-25:
If the printer isn't detected then the Monitor window doesn't populate.
If another application is using the port then Cura can't connect.
If the USB Printing plugin is installed then Cura should connect when it starts up.  I think it gets installed by default.
There have been some people who found that their USB cable was for charging only and had no data capability so Cura couldn't connect.
The USB-SERIAL CH340 driver must be installed in the computer.  With the USB cord plugged into a port on the computer then in Device Manager under "Ports (COM & LPT)" you should see something like "USB-SERIAL CH340 (Com6)".  Try different USB ports.  Sometimes the driver will only be installed on a specific physical port.  That's a Windows thing.

fvrmr, 2021-03-26:
Hi @bryceOctovioOstrem can you check if the suggestion of @GregValiant works for you?
We don't maintain USB printing because our printers don't use USB printing anymore.

closed 2021-04-09 by no-response[bot]

Ghostkeeper, 2021-04-12:
I guess it would be handy if it says something like "Your printer is not connected" on that page, right?

GregValiant, 2021-04-12:
@Ghostkeeper,  Regarding the Monitor (and ignoring USB printing)
It would be nice to have features like a response textbox to note the printer responses to commands like M503, a connect/disconnect button, a checkbox to "connect at startup", and maybe the ability to make adjustments to the baud rate.  I'm not talking about duplicating PrintRun, but some communication functionality beyond the simple box to send a command.

(I read a comment from @fieldOfView that sounded like he is taking a new look at USB Printing.  So there may be changes in store for the Monitor page anyway?)

Ghostkeeper, 2021-04-12:
Those are some features that other people have requested before too.

> like a response textbox to note the printer responses to commands like M503

https://github.com/Ultimaker/Cura/issues/7565

> a connect/disconnect button, a checkbox to "connect at startup", and maybe the ability to make adjustments to the baud rate.

These could conceivably fall under https://github.com/Ultimaker/Cura/issues/3449 , although the "connect at startup" checkbox wasn't mentioned before as far as I know. The feature to select which device/COMport to connect to is really common.

All of these feature requests suffer the same fate. Since Ultimaker printers don't support USB printing any more since the Ultimaker Original, they don't want to spend time on such features. I could develop such things in my spare time, but I don't have a USB-capable printer either (and it wouldn't really be my favourite weekend time expenditure anyway). We're basically at the mercy of external developers to develop this, and it wouldn't be a small or easy feature to add to Cura for them.

GregValiant, 2021-04-13:
That's what I figured.  Thanks for the response.
I jump back and forth between the UI I created for the printer and Cura's USB Printing function.  Having to wait for the Marketplace to load, selecting USB Printing, Quitting Cura, and then re-starting Cura and then do it all again to turn USB Printing off takes a while.  I don't use USB printing, but a lot of questions come up.

commit 4fc91d4f (2021-04-12): Add notification to connect the printer even if the printer is not network-connected

There could be other methods of connecting than the network connectivity of the UM3NetworkPrinting plug-in. For instance, perhaps other network connection plug-ins can provide information in this monitor stage. Or the user might want to use a USB cable.
I'm explicitly quite vague about how it should be connected because this label should work for all types of connections except UM3NetworkPrinting.

Fixes #9444.