Ultimaker/Cura #9445: Some way to measure in Cura

xrbeattx, 2021-03-24:
### Is your feature request related to a problem?

I print an old design from CURA because it looks exactly like my old design but I cant tell the difference because their is no way to measure in CURA. "Just use the squares" NOPE, dosent help and is a lazy excuse. "Import a ruler and use that" NOOOOOOOO, just please add a measuring tool! 

### Describe the solution you'd like

Copy the measuring tool In SW

### Describe alternatives you've considered

Killing myself

### Affected users and/or printers

Everyone

fieldOfView, 2021-03-25:
I am planning to finish up my Measure Tool plugin and submit it to the Marketplace in april. It is currently in a "proof of concept" state, not ready for consumption: https://github.com/fieldOfView/Cura-MeasureTool/

fvrmr, 2021-03-25:
Hi @xrbeattx thank you for your feature request. 
Good to know that @fieldOfView is working on a plugin!
This is a duplicate of https://github.com/Ultimaker/Cura/issues/2545 

The alternatives you've considered is also pretty serious, so please don't joke about it!

Will close this issue since it is a duplicate.

closed 2021-03-25 by fvrmr

fieldOfView, 2021-04-08:
There's a test-version with minimal functionality available here:
https://community.ultimaker.com/topic/36573-measure-tool-plugin/