Ultimaker/Cura #9446: Support XY distance gets applied to support roof

Wdfitzpat007, 2021-03-24:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Enable support roof

Eliminate Z Separation

Slice

Inspect support roof

### Actual results

Support roof is not touching in some areas

In those areas the roof is spaced from the model

### Expected results

Support roof is completely touching the bottom of the model.

From the title you can see the conclusion I came to. When you look layer by layer you can see an XY separation on the roof. In some areas this results in a Z separation which clearly is not the intended behavior. This will go a long way to preproducing the more difficult details on FDM.

Wdfitzpat007, 2021-03-24:
One solution would be to calculate the roof XY offset separately and give it its own value.

fvrmr, 2021-03-25:
Hi @Wdfitzpat007 thank you for your bug report. 
Could you provide me a project file where it is happening?
Thank you in advance!

Wdfitzpat007, 2021-03-25:
> Hi @Wdfitzpat007 thank you for your bug report.
> Could you provide me a project file where it is happening?
> Thank you in advance!

Absolutely, one moment

Wdfitzpat007, 2021-03-25:
> Hi @Wdfitzpat007 thank you for your bug report.
> Could you provide me a project file where it is happening?
> Thank you in advance!

The xy gap does appear to be different than the normal gap upon further inspection. However there is still a gap which is not the intention, and I would like control of that gap as my bottom surfaces are still not as sharp as my top surfaces.
[New folder (5).zip](https://github.com/Ultimaker/Cura/files/6205239/New.folder.5.zip)

fvrmr, 2021-03-26:
Thank you for your project file. If you change the `minimum support x/y distance` to 0 mm the gap is gone.
[image]
Let me know if that works for you!

Wdfitzpat007, 2021-03-26:
Yup, saw minimum xy, but since minimum implies that there is a gradient and the regular xy offset does not imply a gradient. I thought it was irrelevant.

Suggestions:

Make the naming system cohesive by changing:

"Support XY distance"
To
 "Maximum Support XY distance"

The user then immediately understands what minimum support distance is. Also since Z override XY is a discreet variable, Maximum would be truthful as it's the maximum until overridden.

Alternatively, and I like this idea much less, changing:

 "minimum support XY distance"

To any of:

Support Bond xy offset
Support Attachment xy offset
Support Weld XY offset

Would clarify what the variable does. However, the affomentioned change makes more sense.

Ghostkeeper, 2021-04-12:
"Maximum Support X/Y Distance" isn't quite clear either. If the Support Distance Priority setting is set to "X/Y Overrides Z" then the Support X/Y Distance setting is really the minimum distance that support needs to keep from your model. If it's set to "Z Overrides X/Y" then it indicates the minimum distance unless there is a surface nearby in the Z direction. The support may still be farther away from the model in both cases.

The Settings Guide may clear up the precise behaviour of these three settings: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_xy_distance.md#xy-versus-z-distance

I also don't think that something like "Support Bond X/Y Offset" would clarify to the user what the setting does at all.

closed 2024-12-08 by github-actions[bot]