Ultimaker/Cura #9447: Center the result on buildplate when using "Arrange all models"

printingotb, 2021-03-25:
### Is your feature request related to a problem?

When using Arrange all models, the result is often offset from the centre of the buildplate. Some printbeds may be colder at the edges, resulting in poor bed adhesion. 
It also just feels better having them in the center :)

### Describe the solution you'd like

When arranging all models, also center the resulting "group" (but do not center the z-height when drop to buildplate is disabled).

### Describe alternatives you've considered

Add it as an option in preferences.

### Affected users and/or printers

Everyone

Before arranging:
[image]
After arranging:
[image]

fvrmr, 2021-03-25:
Hi @printingotb thank you for you feature request.
I will discuss this with the team.
Keep you posted.

fvrmr, 2021-03-26:
I  have discussed it with the team. But we decided to defer this issue, the reason why is because this would only work for printers without disallowed area's and since Ultimaker printers do have area's that are disallowed it is not a priority for us. 
Hope you understand!

printingotb, 2021-03-26:
I see. Thank you for looking into it :smiley:

Ghostkeeper, 2021-04-13:
This problem really arises from a change we made in the arranger: https://github.com/Ultimaker/libnest2d/commit/31391fd173249ad9b906390058e13b09238fadc8

The problem originates from Cura's feature of having disallowed areas in the build volume. The arranger originally can't handle non-convex spaces to place things in, but it does allow items to be "fixed", so they won't get moved by the arranging. So we made the disallowed areas "fixed" items. But then other items are still being grouped together with those fixed items. So what we did instead is to place the first item in the middle to allow other items to join together with that one. This does mean that one item always gets placed in the middle, and other items get placed around that one. That's not ideal and in your case makes it not centred.

It could also prevent the arranging from being successful even though objects could easily be placed side by side as long as they are not in the middle.

printingotb, 2021-04-14:
Could you add the arranged items to an actual group (as with the group models command), then center the group as an object? (And then un-group it if it wasn't a group already).
This seems to center them.

Ghostkeeper, 2021-05-03:
It would still not work if the second item doesn't fit on the build plate then. If it doesn't fit, the arranger doesn't arrange it in the same scene (it creates a second build plate and places it in the middle of that one).

printingotb, 2023-06-29:
Still think this is relevant :thumbsup: