Ultimaker/Cura #9448: Change dialogue wording Model to Group when working with groups.

printingotb, 2021-03-25:
### Is your feature request related to a problem?

Not really a problem, it would just feel more intuitive.

### Describe the solution you'd like

Change "Center Selected model" to Center Selected Group" and so on.

### Describe alternatives you've considered

None

### Affected users and/or printers

everyone

(Sorry for the ugly picture, couldn't print screen the dialogue)
[image]

fvrmr, 2021-03-25:
Hi @printingotb thank you for your feature request.
I will also discuss this with the team. Keep you posted!

rburema, 2021-03-26:
Mad a commit that fixes this (or more accurately, works around it). Should be in 4.9, since we didn't have string-freeze yet :-)

We decided it was easier (also for translations) to just make it 'Center/Delete/Multiply Selected' instead. Hope that addresses the problem!

closed 2021-03-26 by rburema

printingotb, 2021-03-26:
That works, thanks!

commit 729d3afd (2021-03-26): Workaround discinction model/models/group.

Fixes #9448