Ultimaker/Cura #9449: Change wording for Feature request instruction

printingotb, 2021-03-25:
### Is your feature request related to a problem?

Not really a problem, a tiny bit confusing.

### Describe the solution you'd like

Change the instructional text "Again, if possible, think about why these alternatives are not working out."
I can't find what instruction "Again" is referring to?

### Describe alternatives you've considered

Spend my time on more important stuff ha!

### Affected users and/or printers

Those with a sharp eye

[image]

fvrmr, 2021-03-25:
Hi @printingotb thank you for your sharp eye. 👁️ 
The again is referring to _if possible_, because it was mentioned at the previous header. And _not working out_ is referring to the alternatives that doesn't work.
But I also think that removing **Again** is better for that sentence! So I will change it. 😄

printingotb, 2021-03-25:
Excellent :partying_face:

commit be638955 (2021-04-12): Remove superfluous 'Again' phrasing

It sets the wrong tone.

Fixes #9449.
closed 2021-04-12 by Ghostkeeper