Ultimaker/Cura #9454: Really difficult to get sections of print to be 100% infill. Also settings don't load.

Sensebellum, 2021-03-26:
### Application Version

Really difficult to get sections of print to be 100% infill. Also settings don't load.

### Platform

Really difficult to get sections of print to be 100% infill. Also settings don't load.

### Printer

Really difficult to get sections of print to be 100% infill. Also settings don't load.

### Reproduction steps

Really difficult to get sections of print to be 100% infill. Also settings don't load.

### Actual results

Really difficult to get sections of print to be 100% infill. Also settings don't load.

### Expected results

Really difficult to get sections of print to be 100% infill. Also settings don't load.

Really difficult to get sections of print to be 100% infill. Also settings don't load.

StuSerious, 2021-03-26:
gosh.. people are getting creative even with forms to not fill or provide any info whatsoever 🤣

fvrmr, 2021-03-26:
Hi @Sensebellum please fill in the form correctly, otherwise we can't help you properly. 
Thank you in advance.

Sensebellum, 2021-03-29:
Great work!

closed 2021-03-29 by Sensebellum