Ultimaker/Cura #9455: Cura does not show any printers connected (printers don't connect to Cura)

bestadamdagoat, 2021-03-26:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Custom (da Vinci mini w+)

### Reproduction steps

1. I import an object into Cura
2. I slice the object

### Actual results

I can't print over usb

### Expected results

I should be able to print over usb

[image]

[cura.log](https://github.com/Ultimaker/Cura/files/6209473/cura.log)

fvrmr, 2021-03-26:
Hi @bestadamdagoat thank you for your bug report.

See here what you can try to make it work https://github.com/Ultimaker/Cura/issues/9444#issuecomment-806950008
Let me know if this works for you.

We don't maintain USB printing anymore since our printers don't use USB printing.

bestadamdagoat, 2021-03-27:
I don't know what to do in that comment, I have closed all applications that could be interfering with the port, heck I even changed the port from COM3 to COM6 and switched through all my USB ports and it still does not work. 

I don't see USB-SERIAL CH340 (Com6) in device manager, but it shows my printer's name with (COM3) next to it, which I later changed to COM6 in hopes of fixing it. How do I check if the USB printing plugin successfully installed?

GregValiant, 2021-04-10:
In the _Marketplace_ under _Installed_ - scroll down to _USB Printing_ and make sure it is ticked.

- Quit Cura.
- Unplug the USB cable from either the printer or the computer.
- Power off the printer.
- Power on the printer.  Wait for it to boot up.
- Plug in the USB cable.  The printer should re-boot when it sees the 5volt signal.  This is hardwired into most mainboards to accomplish an external reset.
- Start Cura.  The printer may re-boot again if/when Cura opens the port.

The post @fvrmr linked and the above is all I've got.  If it still doesn't work I don't know what it could be.

Ghostkeeper, 2021-04-12:
The USB Printing plug-in is definitely enabled. In the log file it produces this error message over and over:

```
2021-03-25 23:48:27,648 - DEBUG - [Thread-2] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM7 if baud rate 38400 works. Retry nr: 0. Wait timeout: 3
2021-03-25 23:48:27,651 - WARNING - [Thread-2] UM.Logger.logException [106]: Exception: Unable to create serial
2021-03-25 23:48:27,653 - WARNING - [Thread-2] UM.Logger.logException [110]: Traceback (most recent call last):
2021-03-25 23:48:27,655 - WARNING - [Thread-2] UM.Logger.logException [110]:   File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\USBPrinting\AutoDetectBaudJob.py", line 54, in run
2021-03-25 23:48:27,656 - WARNING - [Thread-2] UM.Logger.logException [110]:     serial = Serial(str(self._serial_port), baud_rate, timeout = read_timeout, writeTimeout = write_timeout)
2021-03-25 23:48:27,658 - WARNING - [Thread-2] UM.Logger.logException [110]:   File "serial\serialwin32.py", line 31, in __init__
2021-03-25 23:48:27,660 - WARNING - [Thread-2] UM.Logger.logException [110]:   File "serial\serialutil.py", line 240, in __init__
2021-03-25 23:48:27,662 - WARNING - [Thread-2] UM.Logger.logException [110]:   File "serial\serialwin32.py", line 62, in open
2021-03-25 23:48:27,664 - WARNING - [Thread-2] UM.Logger.logException [110]: serial.serialutil.SerialException: could not open port 'COM7': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
```

You say you have a custom printer. Do you know which baud rate it expects to communicate at?

bestadamdagoat, 2022-08-26:
Sorry for the necropost, needed to close some of my issues. I decided to get a new printer (specifically the Ender 3 V2 w/ a Raspberry Pi 3 running OctoPrint) because daVinci does the following:
- Makes it near impossible to modify your printer
- Focuses on design rather than repairability
- Creates issues like this to force you to use their glitchy closed-source software
- Forces you to use their filament through built in chips justifying it by "giving enhanced safety features and tracking filament usage." Don't know if this was because of misuse but I used third-party filament by using their "unlocker" add on and ended up having the printer quit mid-print due to a jam and because it couldn't read temperatures. Turns out the thermistor (might be) broke and replacing it is very complicated.
- Has a whole jail breaking community and patches different methods

For any people looking for a solution in the future, I suggest recycling, reselling, or donating (as the printer is good for children w/ adult supervision due to how hard it is to mess up something when doing it their way) the printer and replacing it with an Ender 3 V2/Pro, Prusa, or any other printer that is supported by OctoPrint.

closed 2022-08-26 by bestadamdagoat