Ultimaker/Cura #9456: Cura 4.5 crashes on my work pc

jason3211, 2021-03-26:
### Application Version

4.5.0

### Platform

Windows 10

### Printer

Sovol SV01

### Reproduction steps

install cura
attempt to run cura

### Actual results

Splash screen appears and appears to be loading, main screen flashes but immediately closes, icon in tray also dissapears

### Expected results

Cura should startup successfully

fvrmr, 2021-03-26:
Hi @jason3211 thank you for your bug report.
Did you try this also with the latest (4.8) version of Cura?
Can you let me know if that will work?

Ptit-Pain, 2021-03-26:
i have the same problem but i test on cura 4.8.0 and cura 4.6.0, there is the same problem

jason3211, 2021-03-29:
im pretty sure that i figured out what was causing mine.  my work pc has a rx480 in it that i mine on so my gpu was set to compute in the amd software.  changing it back to graphical seems to have fixed the issue

Ptit-Pain, 2021-03-29:
Could you tell me, precisely what did you do to fix the problem because I have an rx560, so I think I have the same problem as you and I don't quite understand what you mean by "changing it back to graphical seems " ?

Ptit-Pain, 2021-03-30:
Please man, i really need ur help

Ghostkeeper, 2021-04-13:
From the log it seems that the application freezes when creating the graphical interface. That's one moment when a lot of things are happening, so it's hard to tell for us. But one thing that happens is that the 3D scene is being rendered for the first time so GPU issues could indeed be the case.

Updating video drivers might help.

> I don't quite understand what you mean by "changing it back to graphical seems " ?

Try these instructions: https://www.amd.com/en/support/kb/faq/dh-024

closed 2024-12-08 by github-actions[bot]