Ultimaker/Cura #9457: Crash on importing new material. ID not saved.

fuzebox1, 2021-03-26:
**Application version**
4.8.0-1

**Platform**
Arch Linux, Intel gpu

**Printer**
Biqu B1

**Reproduction steps**
1. Imported official .xml.fdm_material from add:north. Cura says success.
2. Wait a moment, import a model or change a setting. Cura crashes saying file is read only. Does not specify what file, probably materials file?
3. Same no matter if i select the new profile or not.
4. Same even if i set materials file to RW.
5. Reloaded Curas default settings (that one hurt), still same.

**Expected results**
Cura should save the material profile.

**Log file**
python Traceback:
```
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/cura/Machines/Models/MaterialBrandsModel.py", line 56, in _update
    item = self._createMaterialItem(root_material_id, container_node)
  File "/usr/lib/python3.9/site-packages/cura/Machines/Models/BaseMaterialsModel.py", line 209, in _createMaterialItem
    "is_read_only":         self._container_registry.isReadOnly(metadata["id"]),
  File "/usr/lib/python3.9/site-packages/UM/Settings/ContainerRegistry.py", line 321, in isReadOnly
    return provider.isReadOnly(container_id)
  File "/usr/lib/uranium/plugins/LocalContainerProvider/LocalContainerProvider.py", line 188, in isReadOnly
    file_path = self._id_to_path[container_id]  # If KeyError: We don't know this ID.
KeyError: 'E-PLA_bowden_1.75mm'
```
Tried setting the python files to RW (later reverted), and the materials file to RW. Same result.

Logs did not provide any relevant information, just traceback.

I have submitted the report from within cura se well.

**Additional information**
I changed the materials file to read write (chmod 666), still same issue.

fvrmr, 2021-03-29:
Hi @fuzebox1 thank you for your bug report. 
Are you using the appimage of Cura?
Because it looks like that you have python 3.9 and we use another version of python.

nallath, 2021-03-29:
Might be related to: https://github.com/Ultimaker/Cura/issues/9464

fuzebox1, 2021-03-29:
@fvrmr I did not realize there was a appimage in the aur. I am using the community repo package and it appears to use system python.
I installed cura-appimage-bin from the aur, and it appears after testing to be working as intended now.

closed 2021-03-30 by nallath