Ultimaker/Cura #9458: Cura 4.8 on Ender 3 showing 0 minutes

SimplyLucKey, 2021-03-27:
### Application Version

4.8

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Scaled the print down 
2. Switched preset printer to another printer

### Actual results

1. Nothing
2. Other printer can slice fine

### Expected results

Time does not show 0 minutes

I had a couple of first layer test variations that I have used before, and worked fine. 
However now that I'm trying to slice these same `.stl` file and the time shows 0 minutes and it would not print. 

I have scaled it down. The print area is clearly smaller than the bed. Nothing.
When I switch the preset printer, it sliced fine. When I switch it back, it went back to 0 minutes.

fvrmr, 2021-03-29:
Hi @SimplyLucKey thank you for your bug report.
Can you share a project file (file - save project) with this issue. Because I can't reproduce this right now.
Thank you in advance.

SimplyLucKey, 2021-03-29:
> Hi @SimplyLucKey thank you for your bug report.
> Can you share a project file (file - save project) with this issue. Because I can't reproduce this right now.
> Thank you in advance.

Sure thing. I got two projects for you to look at. 
[bugs.zip](https://github.com/Ultimaker/Cura/files/6221055/bugs.zip)

fieldOfView, 2021-03-29:
Looks like your model has simply become too thin. Either increase the height of the model, decrease the layer height, or change the Slicing Tolerance setting.

SimplyLucKey, 2021-03-30:
> Looks like your model has simply become too thin. Either increase the height of the model, decrease the layer height, or change the Slicing Tolerance setting.

What does slicing tolerance do? exclusive, middle, and inclusive?

How does decreasing my layer height affect the print?

Ghostkeeper, 2021-04-13:
> What does slicing tolerance do? exclusive, middle, and inclusive?

Conceptually, it ensures that the print stays within the bounds of the model (so it's always too small, "exclusive"), or that the print always includes the entire bounds of the model (so it's always too big, "inclusive"), or that the print approaches the surface as good as possible ("middle").
The precise behaviour is quite complex to describe, but there are some diagrams in the Settings Guide plug-in: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/slicing_tolerance.md
What fieldOfView meant, I think, is to set the Slicing Tolerance to "inclusive" such that the print always encapsulates the entire volume of the model, effectively making the print a little bigger.

> How does decreasing my layer height affect the print?

In many ways. Mainly it improves the overall print quality and allows printing of finer details. The latter is what fieldOfView meant to achieve. It also greatly increases the print time and may need adjustment of other settings to get a similar print quality. See also the Settings Guide again: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/resolution/layer_height.md

However I think the real problem here arises from the Use Adaptive Layers setting. The `CE3_X` model is 0.2mm tall. The initial layer height is also set to 0.2mm, so it should slice through the middle at Z=0.1mm and produce a single layer. However Adaptive Layers seems to have trouble with having just a single layer. It normally works by measuring distance between subsequent layers (starting from the default layer height) and determining the optimal layer height based on that. But if you just have a single layer, I can see how that could be going wrong.

fvrmr, 2021-04-16:
Hi @SimplyLucKey we have discussed this with the team and decided to defer this bug. Because it only happens at one single layer of your print, so it is really an edge case. 

Like @Ghostkeeper mentioned you can `disable use adaptive layers.` 
Hope you understand.

closed 2023-06-24 by github-actions[bot]