Ultimaker/Cura #9460: [4.8.0] Infill going through the wall

yngndrw, 2021-03-28:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

E3D Toolchanger

### Reproduction steps

I just loaded my model with my standard settings.

### Actual results

On layer #11, the infill has been placed outside of the model.

### Expected results

The infill should be constrained to the model.

A picture is a thousand words:
[image]

I haven't finished this design or configured the correct settings, I was just looking to see approximately how long it would take to print when I noticed this issue. The model was made using Onshape.

Project file and .stl:
[Mount Base - Test.zip](https://github.com/Ultimaker/Cura/files/6216706/Mount.Base.-.Test.zip)

[cura.log](https://github.com/Ultimaker/Cura/files/6216707/cura.log)

fvrmr, 2021-03-29:
Hi @yngndrw thank you for your bug report. 
I can reproduce this in 4.8 but in our Arachne beta release it doesn't happen. Could you check if there the issue doesn't occur either?
[image]

You can check the Arachne release here: 
https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta

Also I am wondering why you want to print this at `one at a time`? Because you only have one model. 
I will discuss the issue with the team since it is happening on 4.8 
Keep you posted.

yngndrw, 2021-03-29:
Arachne does seem to work correctly, although I would note that it seems to be approx 10-20% slower at slicing. (Non-scientific measurement) The supports are slightly different, especially the little extra bits that tree supports like to add, there's a few single-layer rings that it has placed around the horizontal holes in the lower portion - Although this isn't unique to Arachne)

The Cura profile is just my main PETG profile with some changes from the last thing I printed. I've started using `one at a time` in order to reduce stringing between different parts, although most of the time I just print a single thing.

fvrmr, 2021-03-30:
I have discussed it with the team and this was already an issue before. See https://github.com/Ultimaker/Cura/issues/6728
We have a ticket on our backlog. Developers see CURA-7845

I will close this issue since it is a duplicate.

closed 2021-03-30 by fvrmr