Ultimaker/Cura #9461: "multiple layers at a time" mode

hyansuper, 2021-03-28:
### Is your feature request related to a problem?

"once at a time" mode saves time because extruder don't need to travel between multiple models every layer, but it requires more space between each model.

### Describe the solution you'd like

We can print multiple layers for each model, like 3 or 5 or even 10 layers for each model, before extruder travels to another model. so we save x/y travel time at the expense of a little more z travel time.
and models can be placed close together like "all at once" mode.
This will be very useful when we want to print many identical parts at once, we just wait for the first few layers printed to make sure they stick well to platform, then we can leave the printer and enjoy life

### Describe alternatives you've considered

Alternatives？ maybe manually manipulate gcode.

### Affected users and/or printers

All

GregValiant, 2021-03-29:
This has come up before.  For myself, the layer cooling duct on my Ender is 0.5mm above the part and it can't be moved up (it's already against the heat block silicone insulator).  The most I can really do is 1 layer at a time.  So before this could be implemented a user needs to know the clearance below the print head, and the layer height in order to calculate the possible "layers-at-a-time".  If I was to do 2 layers at 0.2 layer height then I'm pretty sure the duct would start catching on prongs sticking up and on other things.  "Non-planar" printing has the same issues because of the clearance problem.
Adjusting the gcode is possible.  Multiple gcode files that print maybe 5mm each can be spliced together.   The transition code at each splice might need to adjust the Home Offsets in addition to the extruder location.  It's something like overcoming the gantry height limit when printing One-at-a-Time.  It is also tedious and time consuming.

hyansuper, 2021-03-29:
> the layer cooling duct on my Ender is 0.5mm 
do you mean the fan output part?
I didn't notice the problem, but we can arrange print sequence from right to left if the fan output is on the left side to the extruder

fvrmr, 2021-03-29:
Hi @hyansuper thank you for your feature request. 
I will discuss it with the team. Keep you posted!

GregValiant, 2021-03-29:
"...we can arrange print sequence from right to left."
That's part of it.  So long as the print head doesn't interfere with the taller part it should be good.  Moving from front to back for the succeeding rows would work as well.
Consider a support block as large as the build plate, and as tall as the tallest print.  Set the bottom at 5mm and set it as a cutting mesh so nothing prints that is inside the blocker.  Slice.  That's 1 gcode file.  Move the support blocker up 5mm and insert another cutting mesh with the same properties that has it's top at the 5mm height.  Slice.  That's the 2nd gcode file.  Keep moving the 2 blocks up 5mm at a time and produce a gcode file each time for the visible 5mm stripe of part between the upper and lower blockers.  Splice the files together taking due care regarding crashes.  You may or may not need to insert G92 lines at the transition layers (to sync the E-location at each transition.
You could also use M32 and call the sub-files from a master file.  That's when you start getting fancy.

Sorry for talking past you Fenne.

fvrmr, 2021-03-30:
I have discussed it with the team and like @GregValiant it has come up before. See https://github.com/Ultimaker/Cura/issues/6642
We still decided to defer this issue. We believe it has value, but it will take great effort to implement this. Hope you understand!

And your ideas are always welcome @GregValiant 😄

I will close this issue since it is a duplicate. But feel free to continue the thread in #6642

closed 2021-03-30 by fvrmr