Ultimaker/Cura #9463: Printer stops during print

federicodilo, 2021-03-28:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

helloBEEprusa

### Reproduction steps

1. I have started a 2 h+ print
2. Noticed that the printer stopped after at about 1.5 h
3. I went trough monitor tab
4. Pressed pause button
5. ...The page has become blank

### Actual results

The printer stopped after at about 1.5 h

### Expected results

I would have preferred the printing not to stop, or at least to be able to resume it

The log file is here: [cura.log](https://github.com/Ultimaker/Cura/files/6217328/cura.log)

Here some screenshots:
[image] 
[image]
(Please note I use a scale of 150% DPI)

fvrmr, 2021-03-29:
Hi @federicodilo thank you for your bug report. 
Could you tell me at what time and date this has happened? Because your log file is extremely long, so it saves time reading the whole log. 
But scanning your log I saw some things that are related to the connection, so probably your computer doesn't recognizes your printer at some point which might be the problem for stopping the print.

Please be also aware that our printers doesn't use USB printing so we don't maintain this anymore.

Ghostkeeper, 2021-04-13:
This seems to be the source of the issue:

```
2021-03-28 14:56:01,822 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: An unexpected exception occurred while writing to the serial.
2021-03-28 14:56:01,822 - WARNING - [MainThread] UM.Logger.logException [110]: Traceback (most recent call last):
2021-03-28 14:56:01,841 - WARNING - [MainThread] UM.Logger.logException [110]:   File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\USBPrinting\USBPrinterOutputDevice.py", line 247, in _sendCommand
2021-03-28 14:56:01,841 - WARNING - [MainThread] UM.Logger.logException [110]:     self._serial.write(new_command)
2021-03-28 14:56:01,841 - WARNING - [MainThread] UM.Logger.logException [110]:   File "serial\serialwin32.py", line 315, in write
2021-03-28 14:56:01,841 - WARNING - [MainThread] UM.Logger.logException [110]: serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'Accesso negato.', None, 5))
```

So it seems that after 1 hour and 37 minutes, the operating system suddenly said "No, you're not allowed to send anything through this USB cable any more."
It could be that the cable got disconnected? Bad cable or bad connection? Or maybe a firewall was blocking writes while it was updating itself or something?

federicodilo, 2021-04-13:
Thanks for your reply, I haven't had much time for 3D printing lately.

I can tell you it's not the first time this happens, mainly this problem has occurred in prints lasting over an hour.
I haven't touch really anything since the start and I disabled automatic hibernation of the PC from options.

It seems like the board reject to connect to the PC because it is already busy, despite there isn't any other process making requests on USB other than Cura (the computer replies in system's language "access denied").
Do you think it's a problem about the board?

Ghostkeeper, 2021-04-29:
Could be. Maybe it's overheating?

As far as I can see though, there is nothing that we can really do about it from Cura's side if Windows says no.

Ghostkeeper, 2021-05-03:
Sorry, there is nothing that Cura can realistically do to gain access to USB ports if Windows forbids it.

closed 2021-05-03 by Ghostkeeper