Ultimaker/Cura #9464: Crash when saving 3MF file

X-Ryl669, 2021-03-28:
### Application Version

4.8.0

### Platform

Linux

### Printer

SapphirePlus

### Reproduction steps

1. Drop a STL in a Cura
2. Click File / Save project / Save
3. Crash

### Actual results

Crash dialog with:

Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/cura/CuraApplication.py", line 1063, in event
    return super().event(event)
  File "/usr/lib/python3.9/site-packages/UM/Qt/QtApplication.py", line 464, in event
    event._function_event.call()
  File "/usr/lib/python3.9/site-packages/UM/Event.py", line 218, in call
    self._function(*self._args, **self._kwargs)
  File "/usr/lib/python3.9/site-packages/cura/Utils/Threading.py", line 34, in _handle_call
    ico.result = func(*args, **kwargs)
  File "/usr/lib/cura/plugins/3MFWriter/ThreeMFWorkspaceWriter.py", line 57, in write
    self._writeContainerToArchive(container, archive)
  File "/usr/lib/cura/plugins/3MFWriter/ThreeMFWorkspaceWriter.py", line 154, in _writeContainerToArchive
    serialized_data = container.serialize(ignored_metadata_keys = ignore_keys)
  File "/usr/lib/cura/plugins/XmlMaterialProfile/XmlMaterialProfile.py", line 154, in serialize
    builder.start("metadata") # type: ignore
TypeError: start expected 2 arguments, got 1

Logs:
Thread 0x00007fefc8a93640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 2530 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1735 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007fefc9294640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1485 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007fefc9a95640 (most recent call first):
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1359 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007fefb0ff9640 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007fefb1ffb640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 2530 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1735 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap
[... 117 lines cut ...]
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007ff03a7fc640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Current thread 0x00007ff060873740 (most recent call first):
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 396 in _logInfoWidget
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 182 in _createDialog
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 86 in __init__
  File "/usr/bin/cura", line 167 in exceptHook
  File "/usr/lib/python3.9/site-packages/UM/Qt/QtApplication.py", line 378 in exec_
  File "/usr/lib/python3.9/site-packages/cura/CuraApplication.py", line 847 in run
  File "/usr/bin/cura", line 235 in <module>

### Expected results

No crash, file saved

nallath, 2021-03-29:
Are you using the appimage? It seems to be that your version of a library that we use is different from what we expect.

X-Ryl669, 2021-03-29:
I'm using what's in Manjaro's (Archlinux) public package manager. How could I check if it's the appimage ?

nallath, 2021-03-29:
That's the issue then; Those packages are not maintained by us. Please contact the maintainer of those packages. 

The only version of Cura that we support are the appimages, which can be downloaded from this repo (under releases) or on the Ultimaker website.

closed 2021-03-29 by nallath