Ultimaker/Cura #9465: Overhang Wall Fan Speed

CCS86, 2021-03-28:
### Is your feature request related to a problem?

Overhanging walls cup or sag.

### Describe the solution you'd like

New setting "overhang wall fan speed". There is already a detection algorithm for overhanging walls. Putting either a fan multiplier or adder setting onto detected overhanging wall features would greatly improve print quality.

### Describe alternatives you've considered

Running a higher fan speed all the time can result in weaker parts, or layer separation. Running the fan very low, or off, causes a big reduction in overhang quality,

### Affected users and/or printers

All.

This has been discussed at least a few times, but all the issue I have found are closed without a resolution or explanation. In some the request has been over complicated. I think almost everyone would benefit in print quality by running a higher fan speed for overhanging walls.

fvrmr, 2021-03-29:
Hi @CCS86 thank you for you feature request.
This is indeed a duplicate of #5140 
I will discuss this again with the team. Keep you posted!

fvrmr, 2021-03-30:
I have discussed it with the team and we decided to make a ticket for our backlog for this feature request. 
I will close this ticket because it is a duplicate of #5140 so we keep everything at one place. 
Developers see CURA-8134

closed 2021-03-30 by fvrmr