Ultimaker/Cura #9467: Minimum infill area

darkyp, 2021-03-29:
### Application Version

4.8.0

### Platform

Windoes 7

### Printer

Ender 3 Pro

### Reproduction steps

Slice with a specified minimum infill area without Top and Bottom layers

### Actual results

The minimum infill area setting is not respected as (probably) there are no top / bottom infill settings specified (hidden by the UI)

### Expected results

Either have separate infill settings for minimum infill area or do not hide the top / bottom pattern and related settings in case they are used by the engine in conjunction with the minimum infill area

Hello Cura team,

The minimum infill area setting is not respected (though visible and configurable) when there are no top and bottom skin layers. I would like to change the speed of the small infill area but it is not possible. This is easily reproduced based on the description, still, let me know if more information is required.

fvrmr, 2021-03-29:
Hi @darkyp thank you for your bug report. 
Could you share a project file (file - save project) where this issue is happening. In that way it is easy to reproduce.
Thank you in advance.

darkyp, 2021-03-30:
While trying to prepare the project file it turned out my initial assumption might not have been entirely correct or I might have encountered two separate problems.

So here is what I am more or less after:
[image]
- I have the small feature turned on and a small feature speed of 50%. Infill speed at layer 45 (for example) seems to be respected (project file: MinFillArea2_2.3mf).

If I lower the small feature speed (25%) this is the result:
[image]
 - infill speed seems not to be lowered (project file: MinFillArea2_1.3mf).

If the build plate has another object next to the one above and small feature speed is set to 50% then this is the result:
[image]
 - again infill speed seems not to be lowered (project file: MinFillArea1.3mf).

On to my minimum infill area problem. This is the result when I have configured one top layer in order to be able to use the minimum infill area for the stem in the middle:
[image]
 - (project file MinFillArea4.mf3) The minimum infill area is detected and the "top" settings are taken into account. However I do not want a top layer as my infill is 100% anyway and for the larger top base layer (10) I would rather want a normal infill speed just as on layer 9. Sure, this can be worked around with an overlapping mesh with different settings, still I would expect it to be more streamlined. Finally I would expect to have some sort of separate settings for minimum infill area not related to the top/bottom settings being enabled. And anyway currently they do not hide the minimum infill area one when they are not present.

[MinFillArea.zip](https://github.com/Ultimaker/Cura/files/6226631/MinFillArea.zip)

Ghostkeeper, 2021-04-13:
Small Feature Speed works, but it only applies to the walls. Darkyp intends it to change the infill pattern as well.

It says this in the description for Small Feature Max Length and Small Hole Max Size, but not in the description for Small Feature Speed. That is perhaps something that can be improved there. It was designed for small holes (like screw holes) to give them better dimensional accuracy and make them stick better to the build plate.

I would indeed expect that Minimum Infill Area applies just the same when there are no top/bottom layers. I consider this a bug and will propose this to the developers to see if it can be planned in.

fvrmr, 2021-04-16:
Hi @darkyp we have discussed this with the team and decided to defer this issue. A better way is to use a infill mesh instead of fixing this bug. 
Hope you understand.

darkyp, 2021-04-16:
Hello and thank you for the information. I will think on a postprocessing job to do what I need. As a separate note I think that using a concentric infill also obeys the small feature speed but will have to test further.

closed 2023-06-24 by github-actions[bot]