Ultimaker/Cura #9468: Reloading models breaks merged origins

BagelOrb, 2021-03-29:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

1. load in two models made for merging
2. right click > merge
3. adjust models in modeling software
4. reload models in Cura using F5 or the bottom pop up

### Actual results

The two models don't align properly any more.
They are shifted by the difference in the bounding box between the two versions of the models, or something similar.

### Expected results

The two models align properly.

The result should be the same as when doing right click > merge on the newer version of the model.

Workaround: ungroup and regroup models.

Showing bug:
[image]

After ungroup > regroup:
[image]

Models before and after the design change:
[dual_cubes_before.zip](https://github.com/Ultimaker/Cura/files/6222618/dual_cubes_before.zip)
[dual_cubes_after.zip](https://github.com/Ultimaker/Cura/files/6222620/dual_cubes_after.zip)

fvrmr, 2021-03-29:
Hi @BagelOrb thank you for your bug report. 
I am able to reproduce this. I will discuss it with the team. 
Keep you posted!

fvrmr, 2021-03-30:
I have discussed it with the team. But we decided to defer this, since we have more critical bug on our backlog. And don't think a lot of users will have this issue. 
Hope you understand!

BagelOrb, 2021-03-30:
I wasn't expecting otherwise tbh.

I just wanted the issue to be logged for future reference.

closed 2023-06-10 by github-actions[bot]