Ultimaker/Cura #9469: hangup when Slicing

LTOZ, 2021-03-29:
### Application Version

Arachne_engine_beta 

### Platform

Windows 10

### Printer

Creality Cr-10s

### Reproduction steps

Load model, rotate, scale 2000%, slice with setting profile

### Actual results

slicer loads half ways and freezes

### Expected results

model to be sliced so I can preview before printing

[cura.log](https://github.com/Ultimaker/Cura/files/6224757/cura.log)

[image]

LTOZ, 2021-03-29:
It seems to be working fine with standard profiles but when I use a custom profile it sometimes slices or sometimes get hung up. Nothing changes in the profile so not sure why it is so inconsistent between projects.

nallath, 2021-03-30:
The slicing crashed. Could you also share the project file that reproduces it?

GusMcN, 2021-04-04:
Windows 10
Sometimes successfully slices, sometimes doesn't seem to start and other times progress bar gets part way along and then reverts to showing no progress.
[image]
[GuideRail.zip](https://github.com/Ultimaker/Cura/files/6254805/GuideRail.zip)

nallath, 2021-04-06:
> Windows 10
> Sometimes successfully slices, sometimes doesn't seem to start and other times progress bar gets part way along and then reverts to showing no progress.
> [image]
> [GuideRail.zip](https://github.com/Ultimaker/Cura/files/6254805/GuideRail.zip)

Same holds for you; Without a project file there is nothing we can do.

GusMcN, 2021-04-08:
Hi Jaime

I’m a new user and wasn’t actually saving a project. I just loaded the STL file I sent you  and made the changes (visible in the screenshot) Low Quality ,100% infill with a raft using PLA temperatures 200/50 for an Ender 3V2 and attempted to slice.

I’ve packaged it up into a project file which I have attached.

Interestingly enough using the project file the problem seems to have disappeared, so it’s probably of marginal use.

I’ve just checked and I can’t get it to happen without the project file either. 

I’ve probably done something only a newbie would do and we’ll never know.

If it happens again and I can capture it in a project file I’ll be back in touch.

 

All the best

Gus

 

 <https://user-images.githubusercontent.com/69304091/113512954-16b00c00-955f-11eb-8e28-7bfa765f83ab.jpg>

closed 2021-04-13 by no-response[bot]