Ultimaker/Cura #9470: [4.8.0] material_standby_temperature is single-valued

bellzw, 2021-03-30:
### Application Version

4.8.0

### Platform

64-bit Windows 10 on a Lenovo Yoga 14 laptop

### Printer

BIBO Dual

### Reproduction steps

1. set material 1 standby temperature to 100 C
2. set material 2 standby temperature to 175 C
3. startup g-code:
M104 T0 S{material_standby_temperature}
M104 T1 S{material_standby_temperature}
G21 ;metric values
G90 ;absolute positioning
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z2.0 F400 ;move the platform down 2mm
T0
G92 E0
G28
G1 Y0 F1200 E0
G92 E0
M109 T0 S{material_standby_temperature}   ; now wait for E1  to reach temperature
M109 T1 S{material_standby_temperature}   ; now wait for E2  to reach temperature
M109 T{initial_extruder_nr} S{material_print_temperature_layer_0}  ; now set the correct extruder to the correct initial temperature
T{initial_extruder_nr}  ; make sure the extruder for the first layer is selected

The intent is to start each of the extruders heating to the standby temperature while the startup code is executed, and end with the extruder NOT being used first at its standby temperature, and the extruder used first at the its initial print temperature.

### Actual results

Actual g-code:
M104 T0 S100.0
M104 T1 S100.0
G21 ;metric values
G90 ;absolute positioning
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z2.0 F400 ;move the platform down 2mm
T0
G92 E0
G28
G1 Y0 F1200 E0
G92 E0
M109 T0 S100.0   ; now wait for E1  to reach temperature
M109 T1 S100.0   ; now wait for E2  to reach temperature
M109 T0 S200.0  ; now set the correct extruder to the correct initial temperature
T0  ; make sure the extruder for the first layer is selected

### Expected results

I expected the substitution for standby temperature for T1 to be 175, not 100.  
M104 T0 S100.0
M104 T1 S175.0
G21 ;metric values
G90 ;absolute positioning
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z2.0 F400 ;move the platform down 2mm
T0
G92 E0
G28
G1 Y0 F1200 E0
G92 E0
M109 T0 S100.0   ; now wait for E1  to reach temperature
M109 T1 S175.0   ; now wait for E2  to reach temperature
M109 T0 S200.0  ; now set the correct extruder to the correct initial temperature
T0  ; make sure the extruder for the first layer is selected

This file shows the startup code in the machine settings and the standby temperature of material 2.
[image]

3mf and gcode generated for this build
[Yad_for_learning.zip](https://github.com/Ultimaker/Cura/files/6225543/Yad_for_learning.zip)

The essential problem seems to be that material_standby_temperature has the value corresponding to the first extruder that is used and ignores the fact that in a multi-extruder environment, there might be different standby temperatures for different extruders.  material_standby_temperature is unaware of the material being referenced by the g-code.

I suggest checking material_print_temperature_layer_0 for the same problem.  In this particular example, Printing Temperature Initial Layer is identical for the two materials.

fvrmr, 2021-03-30:
Hi @bellzw thank you for your bug report.
I can also reproduce this and also on our Ultimaker printers. So I will discuss it with the team. 
Keep you posted.

bellzw, 2021-03-30:
Thank you.

 

I don't know what's actually in the code, but I would think making this an array indexed by the tool number referenced by the M104 or M109 would solve the problem.

Ghostkeeper, 2021-03-31:
The `{material_standby_temperature}` text gets replaced by the stand-by temperature of the first extruder, if the extruder is not specified. It doesn't automatically detect that it's in a segment of g-code that affects the second extruder or anything.

You can use `{material_standby_temperature, 1}` to refer to the stand-by temperature of the second extruder (0-indexed). If the extruder number is not specified, it defaults to 0 (the first extruder). See [the Settings Guide](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/machine_settings/machine_start_gcode.md) for a more complete description.

This is precisely the sort of solution that you envisioned, I think. So I'll close this issue since the feature already exists :)

closed 2021-03-31 by Ghostkeeper

bellzw, 2021-03-31:
Thank you.  That explains everything.