Ultimaker/Cura #9477: need option Read Only for profiles

khunpoum, 2021-03-30:
### Is your feature request related to a problem?

sometime settings in profile has changed

### Describe the solution you'd like

add option for profile Read Only

### Describe alternatives you've considered

have not)

### Affected users and/or printers

it's affect users whom make they best print settings profile and don't want for Cura overwrite some settings in this profile

fvrmr, 2021-04-02:
Hi @khunpoum thank you for your feature request.
I will discuss this with the team. Keep you posted!

Ghostkeeper, 2021-04-07:
Cura doesn't change profiles unless you changed them by hand, except in one case: reading a project file. I think that's the real reason you are requesting this feature. You want to load the nozzle profile, material profile, quality profile printer profile and customised settings from a project file, but skip the custom profile.

I don't think making a profile read-only is the solution to this UX-wise. For every user that understands the feature there would be 10 people that don't understand why they suddenly can't update the profile any more, or that don't understand why a project file results in different settings if they load it on one computer than when they load it into another.

Discussion is going on about this problem and potential solutions here: https://github.com/Ultimaker/Cura/issues/9000 So I'll close this as a duplicate.

closed 2021-04-07 by Ghostkeeper